TR4W/TR4W #2: Connecting to dxc.ww1r.com:7300 causes crash

ny4i, 2015-05-31:
When I select dxc.ww1r.com:7300 from the drop down in the DX Cluster window, the program crashes right after the connect. I suspect the program is receiving something from the cluster it does not expect. I will check this out with the debugger and see what we are receiving.

n4af, 2015-05-31:
Hi Tom
Yeah , I have never been able to isolate, but there is a problem there

73, Howie ny4a

n4af, 2015-06-01:
can't seem to get it to fail Tom. I have had it running all night.

73, Howie

ny4i, 2015-12-01:
I confirmed this is a bug. It is on the connect. My guess is the banner text is too long on a single line and that is causing an issue. I will investigate more.

ny4i, 2016-02-27:
I tested again and it does not fail. Closing for now...

closed 2016-02-27 by ny4i