TR4W/TR4W #12: Program should recognize UP <n>. Now it only recognizes QSX <freq> and UP1..UP5

ny4i, 2015-12-01:
Many spots are formatted as UP <some number>. That should be handled. 
It may also be useful if there is no number, then in CW assume 1 and in phone assume 5.

n4af, 2015-12-01:
> Many spots are formatted as UP . That should be handled.

R U saying the bandmap should handle it Tom ?
That would require a lot of real estate. Also, it is very rare that
contesters encounter split- maybe phone ??

73, howie

http://n4af.net

ny4i, 2015-12-01:
No extra real estate. The band map handles it now. If you click on a spot
that has qsx 14.040, the program handles it very well. Up 1 was an
oversight as it handles up1.

I agree about contests but I did run across a few stations running split in
CQWW this weekend.

Tom

Principal Solutions Architect
Better Software Solutions, Inc.
727-437-2771

http://n4af.net

n4af, 2015-12-01:
rr. Keep up the good work !

73, Howie

http://n4af.net