TR4W/TR4W #13: Add IC9100 to radio tables

ny4i, 2015-12-01:
This is a harder one as the prior Icom limit was the IC7800. The code in about 20 places used code like IC720..IC7800 to check if the radio type is Icom (rather than a radio kind in the TRadioRecord.

To do this, change the code that uses IC7800 as the top of the array to IC9100 (that will work until Icom comes out with something higher than 9100 but it should hold for a few years.

The 9100 also has some differences in some commands. 

Note that Icom has added firmware support to read the frequency of VFO B. It would be good to add that as part of this, then this could also support the updated 7200, 7300, 7600, 7700, 7800, 7850, 7851 and 9100.

ny4i, 2015-12-01:
I should add that this was quite relevant as on my bench, I have an IC9100 and a K3. So for 2nd radio testing, I need to use the 9100 and I did not want to reset the CI-V address from 7C as I have other programs like SatPC32 that use the native address.

n4af, 2015-12-01:
Hope you are able to figure that out-- problem being we do not have most
radios in house to figure out. I only have K3's, so your having IC9100 is a
bonus.

(now back to my head scratching on main pc)

73, Howie

http://n4af.net

ny4i, 2015-12-01:
I decided to stop and start over. I know it is not a logical progression, but I changed many modules simply to make IC9100 the last item in the rigs array. Therefore, since the code treats the IC7800 as the last item in the Icom enumeration, I am going to keep that as the last one. Yes, the list will make no sense but it is the safest thing to do for now.

ny4i, 2015-12-03:
Most of this works just fine now. I am still testing an issue when the 9100 is Radio 2. The serial port just stops and it looks like the radio stops responding. I am not sure how the program handles that case if it does not hear from the radio in awhile or if there is a heartbeat. I'm also going to switch to try the CI-V port instead of the 9100's USB interface.

ny4i, 2015-12-04:
This has now been tested to within an inch of its life. I ran it though the autospot system which changes both rigs to every spot that comes in. I can also click on spots rapidly and no issues on either radio that I can tell.

I am not sure if it is what fixed it but after reading some resources on the web, I set the timeout on the serial port for the radios. These were not set and this should be.

I would actual suggest we switch this to a very robust serial port component and handle this with events. The code is not very centralized in that the ReadFromCATPort procedure is called many places rather than all inputs being handled in an event as a state machine. But that is for another day. Now that I added the 9100, it will be easy to add the 7851, 7850, 7100, and 7300.

n4af, 2015-12-04:
One 'problem' we may have is lack of users of those radios to extend the
testing of the build.

73, Howie

http://n4af.net

ny4i, 2015-12-04:
Agreed. I was just mentioning they should not be hard to add but only once
we get a user with one to verify it works (I should have added that part).
The change the address trick works. I wonder if we should allow the user to
override the icom address and let it take care of itself.

Tom
Principal Solutions Architect
Better Software Solutions, Inc.
727-437-2771

http://n4af.net

commit 39fc500d (2015-12-04): Fixes #13 Adds support for IC9100, fixes some timeout issues with serial port by setting timeouts
closed 2015-12-04 by ny4i