TR4W/TR4W #17: Support sending CW via Command if the rig supports it.

ny4i, 2015-12-01:
Many radios allow sending CW via a command. If the rig supports it, allow an option to send via CMD. While this can currently be done by hand-coding the command sequences in the macros, a better way would be for the program to send the proper commands without the user having to modify the macros. Radios such as the K3, KX3, Orion 2, Icom 7800, IC9100 and others all support this.

n4af, 2016-01-12:
Tested on my K3's. Works great !

ny4i, 2016-01-12:
Certain radios support sending CW via the CAT commands. Those radios include the Ten Tec Orion 2, Elecraft K3/KX3 and certain Icom radios (IC9100, IC7800, IC7850, IC7200, IC7851, ICOM 7600, ICOM 7700).

With this feature, any CW is automatically sent via the CAT interface. Simply set RADIo ONE CW BY CAT = TRUE and all CW messages are sent via a CAT command over the serial port. You DO NOT need to add special characters to the CW Memories. The program handles this. You can also set RADIO TWO CW BY CAT = TRUE to use this feature on both radios. The same interrupt mechanisms such as hitting ESC will stop sending. Note that some radios will support stopping CW if you hit the radio-conected paddle (K3 is know to do this).

ny4i, 2016-01-14:
This is finished and pushed to the branch.

closed 2016-01-14 by ny4i