TR4W/TR4W #23: CTRL-V, CTRL-C and CTRL-X Accelerator keys do not work in some fields

ny4i, 2015-12-03:
The copy, paste and cut shortcut keys do not work in some fields. But, right clicking on the field and selecting paste works. This should be fixed. I noticed it on the Key Programming fields and in the Command field in the DX Cluster window.

ny4i, 2015-12-03:
In the list of changes for the program, I found this: 
*4.36.3 ENABLE CTRL-V DYNAMIC CFG SWITCHING (NE9U)

It is possible that the CTRL-V is being caught in all cases (or more likely bubbling up to the highest level that handles it).

One issue is that for compatibility with the DOS program I suspect, CTRL-C and CTRL-V are used for other things. A Windows program should really not do that. While it would be great if we could redefine those, if not, then if we ensure that text fields handle the CTRL-[C | V | X], that would be good. 

Of course, the any fields int he main window should not handle those or the program would not get the current CTRL-V and CTRL-C functions.

n4af, 2015-12-03:
Oy, i never even thought of that Tom.
I am going to disable it to see if that is the problem...

http://n4af.net

n4af, 2015-12-04:
It looks like copy/paste are supported by ctrl+shift+c (or v)

It appears we can exploit copy/paste inside a program with 'Clipbrd'
documented here:
http://delphi.about.com/od/vclusing/a/tclipboard.htm

73, Howie

http://n4af.net

ny4i, 2015-12-04:
That seems promising but it appears to be centered around Forms. As much of the code is Win32, I wonder if it will have the same effect.

Tom
Principal Solutions Architect, ITIL®v3 
Better Software Solutions, Inc.
727-437-BSS1 (727-437-2771)
@BSSI_Consulting

ny4i, 2026-04-17:
When porting this to D12/D13, look at changing the accelerators keys for Clear Multsheet (CTRL-C) and Execute Configuration File (CTRL-V). Windows programs should reserve those keys.