TR4W/TR4W #25: DARC needs fixed column cabrillo output

n4af, 2015-12-04:
Hi Howie,
 
unfortunately we can not use white spaces for delineating since the 2nd exchange "DOK" may be present (DL stations) or may not be present (non-DL stations). This results in different word counts in all QSO lines depending on station A and B.
 
The same applies for other contests that support 2 exchanges such as IOTA.
 
It would be great if TR4W could follow Cabrillo's fixed-length column concept. Example files can be found here:
http://www.darc.de/referate/dx/contest/10m/en/example-logs/
 
It would be great if you have a look.  We get many TR4W logs (mostly from Russian stations) every year.
 
73 Chris DK9TN

n4af, 2015-12-04:
Hi Howie,
 
I have tried TR4W v4.43.01 and there is the following issue when exporting a Cabrillo file from a "DARC 10m contest" log:

QSO: 28000 CW 2015-11-23 2049 AB1CD         599 2           DL0TUM        599    1      C12  <-- currently it is like this
QSO: 28000 CW 2015-11-23 2049 AB1CD         599 2           DL0TUM        599    1 C12        <-- rcvd DOK should be like this

ny4i, 2015-12-04:
What am I missing? I see no difference in those two lines (except for the comment after <—

n4af, 2015-12-04:
Formatting of the lines; this better illustrates
QSO: 28000 CW 2015-11-23 2049 AB1CD         599 2           DL0TUM        599    1      C12  <-- currently it is like this
QSO: 28000 CW 2015-11-23 2049 AB1CD         599 2           DL0TUM        599    1 C12        <-- rcvd DOK should be like this

I will need to generate a test cabrillo and check. Would be easier if they held to convention counting whitespace delineated fields RATHER than specific column numbers.

73, Howie

n4af, 2015-12-04:
it is different in the original email. cut n paste into this box ruins it...

ny4i, 2015-12-04:
I’m a mainframe guy at my roots so fixed columns are not strange to me :)
Principal Solutions Architect, ITIL®v3 
Better Software Solutions, Inc.
727-437-BSS1 (727-437-2771)
@BSSI_Consulting

n4af, 2015-12-05:
DK9BN has been notified of build for testing
http://tr4w.net/beta/tr4w_setup_4_43.07.exe

n4af, 2015-12-05:
Positive f/b from DK9BN.
Fix moved to build directory http://tr4w.net/4.43/tr4w_setup_4_43.07.exe

closed 2015-12-05 by n4af