TR4W/TR4W #38: Support MK2R native SO2R functions

ny4i, 2015-12-08:
Submitted by RL6M

n4af, 2015-12-08:
Hi Tom -

Would you also open an issue for Support of YCCC SO2R box ?

Tks, Howie

http://n4af.net

ny4i, 2015-12-08:
Here is a blurp from the MK2R manual:

 microHAM Control Protocol, as a whole or in any subset, is the intellectual property of microHAM, s.r.o. and may not be used by any person in conjunction with any hardware interface or control product ("controller") not manufactured by microHAM. Individuals who have purchased a controller from microHAM, s.r.o or its dealers and distributors are licensed to utilize the following subset of commands only with that controller. Complete documentation is available on request and execution of a non-disclosure agreement to qualified software developers who wish to support microHAM interfaces and controllers.

It looks like we would have to ask them to allow us to implement the protocol in TR4W. I suspect it is for the asking but as they reference a non-disclosure, I am not sure how that would work with an Open Source product.

It can be checked to see their current thoughts on this.

As the YCCC SO2R box is an open protocol, I suspect that will be implemented first.

n4af, 2015-12-08:
Hi tom
Out. Shopping today.
Mk2r at gm0gav

Yccc box at n4af
73, howie

ny4i, 2016-02-28:
Here is a summary of an email I received from Joe, W4TV when I asked about the protocol. Please note that for the simple stuff, Joe says this is documented. For the more advanced stuff, he referenced an NDA. For obvious reasons (such as all our code is public), that would be tougher. If we can not get by with the basic commands, then we could figure out a way to do a binary implementation of the more advanced stuff and just distribute the DCU binary and keep the source to that private.

> The basic focus and antenna switch or band data controls are documented
> in the Appendix of the MK2R+ Users Manual and micro2R Users Manual. Both are available in the download area or the interface specific area
> of the web site (www.microHAM-USA.com).
> 
> The primary interface are ASCII commands sent on a virtual serial port
> (the MK2R+ or micro2R "Control" port).  Here is basic information from
> the Users Manual for sending those commands as well as TX focus, RX
> focus and ACC (Band Data or antenna relay):
> 
> All commands are ASCII and are case sensitive. Each command is
> uppercase and ends with <CR>. In this document parameters
> (arguments) for the command are shown in lower case. Router responds
> to any invalid command with an error message where the first
> character is an E is followed by copy of the received command – e.g.,
> Ecommand<CR>.
> 
> Focus control: These commands control transmit focus and headphone
> switching (receive focus).
>   SetTxFocus(): active only when TX Auto Focus has been selected
>                 FT1<CR> SetTxFocus = R1
>                 FT2<CR> SetTxFocus = R2
>   SetRxFocus(): active only when RX Auto Focus has been selected
>                 FR1<CR> SetRxFocus = R1
>                 FR2<CR> SetRxFocus = R2
>                 FRS<CR> SetRxFocus = STEREO
> ACC output control: The ACC output (Band Data or Antenna Relay) can
> be one of 16 values represented by a 4-bit value. The output is set
> using a two digit value between 00 and 16.
>                 AS1dd<CR> SetAccOutputSelection (R1, selection)
>                 AS2dd<CR> SetAccOutputSelection (R2, selection)
> 
> When using micro2R - because it contains no CAT interface - it is wise
> to set the "keyer mode" for each radio when the radio's mode changes:
> 
> Keyer mode control: This command sets the Keyer mode (operating
> > mode) for each radio.
>                 K1M<CR> SetKeyerMode(R1, selection)
>                 K2M<CR> SetKeyerMode(R2, selection)
>                 Mode values are C=CW, V=VOICE, F=FSK.
> 
> micro2R (unlike MK2R+) provides no focus control (TX audio or RX audio
> switching) for audio based (AFSK, PSK, etc.) digital modes.  However,
> it does provide PTT on the respective FSK port (UART) so TX Focus
> should be provided at least in RTTY (FSK) operation.
> 
> Note: supported functions different (because the capabilities are very
> different) between MK2R+ and micro2R but the basic focus controls are
> the same.  I *DO NOT* recommend using the microHAM DVK or CW memories;
> those functions are best integrated with the logging program to retain
> proper control - including issues of timing in "dueling CQs", etc.  The
> more advanced features of MK2R+ - more complex receive audio selection
> and additional support/controls, including the ability to select
> separate TX/RX antennas, etc. through Station Master or Station Master
> Deluxe connected to MK2R+ is in a constant state of development but
> that level of documentation is only available through the microHAM
> factory under NDA (contact microham@microham.com).
>

gm0gav, 2016-02-28:
>     uppercase and ends with . In this document parameters
>     FT1 SetTxFocus = R1
>     FT2 SetTxFocus = R2
>     FR1 SetRxFocus = R1
>     FR2 SetRxFocus = R2
>     FRS SetRxFocus = STEREO
>     AS1dd SetAccOutputSelection (R1, selection)
>     AS2dd SetAccOutputSelection (R2, selection)
>         K1M SetKeyerMode(R1, selection)
>         K2M SetKeyerMode(R2, selection)
>     <mailto:microham@microham.com>).

Tom,
I would suggest that both the Microham & OTRSP start life in their most 
basic form : TX focus & Stereo, that's enough to start using the 
Microham boxes in a basic manner and mimic what the LPT currently does. 
Plus the mode switching for the u2r. Suspect OTRSP will be the same.  
BTW I use a u2r here which I currently use on LPT.

Perhaps You, Howie and me have a skype chat through this one one night?

73 Gav

n4af, 2016-02-29:
I agree 100% that we start basic.
Getting a 'stake in the ground' is what counts at this point.

At some point, we need to set aside ALL requests for enhancements and start
working on introducing
SO2R box support.
NY4I introduction of CWBC shows it can be done.

I have the YCCC box in-house, so that should logically be assigned to me.

73, Howie

http://n4af.net

ny4i, 2016-02-29:
> <mailto:microham@microham.com <microham@microham.com>>).

I have an mk2 so while it not really an SO2R device, I can make sure they
talk.

I would like to add the serial port code using some newer code than raw
Win32.

We should sketch out the features we want and how to integrate it into the
code.

Tom

Principal Solutions Architect
Better Software Solutions, Inc.
727-437-2771

http://n4af.net

> Tom,
> I would suggest that both the Microham & OTRSP start life in their most
> basic form : TX focus & Stereo, that's enough to start using the
> Microham boxes in a basic manner and mimic what the LPT currently does.
> Plus the mode switching for the u2r. Suspect OTRSP will be the same.
> BTW I use a u2r here which I currently use on LPT.
>
> Perhaps You, Howie and me have a skype chat through this one one night?
>
> 73 Gav
>
> >
> > Here is a summary of an email I received from Joe, W4TV when I asked
> > about the protocol. Please note that for the simple stuff, Joe says
> > this is documented. For the more advanced stuff, he referenced an NDA.
> > For obvious reasons (such as all our code is public), that would be
> > tougher. If we can not get by with the basic commands, then we could
> > figure out a way to do a binary implementation of the more advanced
> > stuff and just distribute the DCU binary and keep the source to that
> > private.
> >
> > The basic focus and antenna switch or band data controls are
> > documented
> > in the Appendix of the MK2R+ Users Manual and micro2R Users
> > Manual. Both are available in the download area or the interface
> > specific area
> > of the web site (www.microHAM-USA.com <http://www.microham-usa.com> <
http://www.microHAM-USA.com <http://www.microham-usa.com>>).