TR4W/TR4W #49: SHOWFILTER using dxc.ve7cc.net produces page faul/

ny4i, 2015-12-08:


ny4i, 2015-12-19:
I could not make it fail on the older version. What did the filter need to be to cause it to fail? I suspect the filter string needed to be long.

n4af, 2015-12-19:
Hi Tom -

Exactly so.
It does not need to be 'show filter'; anything that produces a long string
will cause the program check as the listbox field overflows.
It is now limited to 120 characters.
73, Howie
if (TelnetBuffer[c] <= #13)   *or (d > 120)*  then   //n4af 4.43.9 line#750
uTelnet
      begin
       d := 0;

http://n4af.net

n4af, 2017-07-11:
this can probably be closed by 4.43.9 prevention of > 120 char.

ny4i, 2020-07-05:
This works on an AR/Cluster system (plenty long)...

```
NY4I de NC7J 05-Jul 0556Z arc6>
sh dx filter
DX configuration options: 
   Count: 20
   Filter: cont = na and (call not like */p and call not like */b) and (band = 40 or band = 20 or band = 15 or band = 10 or band = 6) and ((skimmer and not skimdupe and (spotter = k2db-# or spotter = k9tm-4-# or spotter = wa2zyu-# or spotter = w4ax-#)) or (not skimmer and (spotterstate = fl or spotterstate = ga or spotterstate = sc or spotterstate = nc or spotterstate = al)))
   Mode: FILTER
   Output: ON
   Comment Options: LOTW Section
```

I also tried on ve7cc.net and it did truncate the output.  It might be nicer to split that but we can revisit that if it comes up again and maybe we can switch to an Indy telnet client.

closed 2020-07-05 by ny4i