TR4W/TR4W #50: AutoCQ fails when telnet spots are heavy

ny4i, 2015-12-08:
Reported by VE5ZX

n4af, 2015-12-14:
I think Gavin may already be investigating a change in how spots are handled ?

"I had a look at the code I added to the bandmap display and it could be slimmed down. Plus limit the bandmap refresh to perhaps every second, when it is being triggered by incoming telnet.  That should clear the issue."

n4af, 2015-12-23:
4.43.10 attempts to remedy- altho it should be noted this is very much a trade off.
My personal pref would be to increase filtering.

TDXSPOTLIST.ADDSPOT now has a sleep for 20 ms. added to perhaps allow autocq to run...

ny4i, 2015-12-23:
There was an idea about adding a timer to do the update once a second rather than upon every spot input. Would that make it better than adding a wait? I hesistant to add waits as the radio polling does that too in several places (well for Icom radios, anyway).

Thoughts?

Tom
Principal Solutions Architect, ITIL®v3 
Better Software Solutions, Inc.
727-437-BSS1 (727-437-2771)
@BSSI_Consulting

n4af, 2015-12-23:
Hi Tom -

Sure. Do you have any code to go with the timer pop ?

I am guessing that we should hold your changes til after 4.44, but your
call.

73, Howie

http://n4af.net

ny4i, 2015-12-23:
The Delphi way (and I see no reason to not do this the Delphi way versus the Win32 way), is to create a TTimer object. Normally, you would just drop a TTimer on a form and set some properties. As these forms are created dynamically via Win32, here is another way to do it at run-time...

In uTelnet.pas, make the following changes...

Add extctrls to the Uses clause

Add this class declaration under TButton (around line 72):

```delphi
TEventHandlers = class
    procedure OnBandMapTimer(Sender : TObject);
  end;
```

Add this declaration around line 101:

```delphi
var
  BandMapTimer                         : TTimer;
  EventHandlers                         : TEventHandlers;
```

Around limne 232, add this right before the TelnetWndDlgProc...
```delphi
procedure TEventHandlers.OnBandMapTimer(Sender : TObject);
begin
   AddStringToTelnetConsole(PChar('Timer fired'), tstTR4W);
   DisplayBandMap;
end;
```

In the WM_INITDIALOGF case add this (around line 413...)

```delphi
 // Create Timer
        if not Assigned(BandMapTimer) then
           begin
           EventHandlers := TEventHandlers.Create();
           BandMapTimer := TTimer.Create(nil);
           BandMapTimer.Enabled := false;
           BandMapTimer.Interval := 1000; // 1 second
           BandMapTimer.OnTimer := EventHandlers.OnBandMapTimer;
           end;
```
Make the WM_CLOSE case look like this (around lime 535)

```delphi
WM_CLOSE: 1:
      begin
        CloseTR4WWindow(tw_TELNETWINDOW_INDEX);
        BandMapTimer.Enabled := false;
        BandMapTimer.Free;
        BandMapTimer := nil;
      end;
```

Add this as the very last line of  ConnectToTelnetCluster (line 630)

```delphi
BandMapTimer.Enabled := true;
```
Add this as the last line of Disconnect at line 647:
```delphi
BandMapTimer.Enabled := false;
```

In tCreateAndAddNewSpot, comment out   //DisplayBandMap; at line 1213.

That is it. Connect to the cluster, and you will see the debug messages fire and the bandmap update once a second...

I tested this and it works. It does look a little choppy in that my eye notices the update interval. Maybe 500 ms would be better. You can play with it and see how it looks. It might be a perception issue as I was watching the Timer Fired messages too.

Adding extctrls should add some MB to the EXE file.

ny4i, 2016-02-05:
This one should now be closed as a duplicate, right?

gm0gav, 2016-02-05:
Yes, I think this can be closed. I am confident it is fixed.

closed 2016-02-05 by gm0gav

n4af, 2016-02-05:
Hi Tom -

I think so.

73, Howie

http://n4af.net

ny4i, 2016-02-05:
Looks like Gav already closed it.

Thanks,

Tom
Principal Solutions Architect, ITIL®v3 
Better Software Solutions, Inc.
727-437-BSS1 (727-437-2771)
@BSSI_Consulting