TR4W/TR4W #53: Cannot enter DX Cluster hostname with a dash (-)

ny4i, 2015-12-08:
When typing a host address into DX Cluster window, if the host name has a dash (-), the program intercepts this and wants to change the XMIT frequency. This makes it impossible to enter the host.

n4af, 2015-12-23:
There are two problems:
1) Dash unfortunately has the same wParam as Alt-N. This can easily be changed to ALT-O and the dash problem will no longer occur. Pls advise if you wish it changed.
2) Entering dash is a problem. 
Key entry is not specific to a window and dash is recognized as a control character. 
Rather than attempt a re-write of wm_KeyDown, under the rare occurance of needing cluster name with a dash, I would suggest just editing TRCLUSTER.DAT and adding the cluster to the list.

n4af, 2015-12-23:
Dash problem #1 picked up in 4.43.10...

ny4i, 2020-07-05:
Yes this is still an issue due to the global input handler in uInputQuery.  It would be possible to add some logic to check what control issued the - and if the host box, do not do anything but pass the - to the window but better to push this for consideration if we cannot a better Delphi forms engine in place with Delphi 10.

ny4i, 2024-02-06:
This now works. I recall I did some work to allow using a dash character. I reworked the way split worked with - as an accelerator if I recall. I tested it and it works. So this will be closed as now working.

closed 2024-02-06 by ny4i