TR4W/TR4W #61: Support of YCCC SO2R box

ny4i, 2015-12-08:


ny4i, 2015-12-08:
Documentation: http://www.k1xm.org/OTRSP/OTRSP_Protocol.pdf

gm0gav, 2016-02-16:
Yes, this is a priority. As is Microham SO2R support, both should prpbably be done at same time

ny4i, 2016-04-13:
Adding a comment from a user to show more interest...

> On Wed, Apr 13, 2016 at 1:00 PM, Allan Taylor <radio.k7gt@gmail.com> wrote:
> I am exploring SO2R and just picked up a YCCC SO2R+ box. I would
> like to use TR4W with it and would like to contact others who may
> be using that combination. The documentation for the box says it
> works with TRlog for Linux but that is all it says re TRxxxx. I will be
> using it with a FT1000 or FT1000MP pair.
> 
> — 
> 73  Allan  K7GT

ny4i, 2020-08-06:
Reference device to test OTRSP commands without an actual device: https://k1xm.org/OTRSP/OTRSP_Reference_Device_Installer.exe

ny4i, 2022-04-22:
During Kyle AA0Z's presentation there was anther request to support this interface. So bump this up the list to a release in a few months. Testing may be an issue but code it to the spec and see if there is a simulator for development.

n4af, 2022-04-22:
If it helps, i can send u yccc box...

ny4i, 2022-04-22:
Let me take a crack at the code first. I’ll let you know when it’s ready to test. You can try and if I don’t get it right, we can go to that step. 

Tom

Principal Solutions Architect
Better Software Solutions, Inc. 
727-437-2771

ny4i, 2022-04-23:
I learned something new today. My Flex supports OTRSP.

"All Flex 6000 rigs implement the OTRSP protocol "

I have a box to implement this with then. I will add this when I add the networking interface for the Flex next month.

ny4i, 2022-04-26:
I have the test program running and can send commands to it. I believe I need some help to go from the OTRSP commands documented here https://k1xm.org/OTRSP/OTRSP_Protocol.pdf and know when each command in TR4W should send what command.

n4af, 2022-04-26:
Hi Tom -

So basically, this
<https://www.k1xm.org/OTRSP/OTRSP_Reference_Device_Installer.exe> runs and
you need interpretation of what actions it triggers in TR4W ?

n4af, 2022-04-26:
i have asked na8v and n5aw to provide some input.  
Using winkey to do the switching works perfect for me, but d/n provide switching for fone.

The RX switching is between otrsp and hardware (audio input).

TR4W major control is via  SwapRadios in LogSubs1.pas

I will take a closer look today...

n4af, 2022-04-26:
> A lot of the commands can be skipped for now.
> REQUIRED:
>   The *TX1, TX2, RX1, RX2*, *RX1S, RX2S*, RX1R, and RX2R commands. RX1R
> and RX2R can be treated as RX1S and RX2S. RTS as a PTT control if
> appropriate.
> I am guessing Mode would be nice, but suspect TR4W can already control
> that based on radio physical settings ?

ok, so some basic checking. if we do not issue MAP, then R1/R2 are defined
by port assignment.

So TX1 command checks if active radio is 2 then SWAPRADIOS
  else
    If ActiveRadio is already 1 then
     exit
------------
The RX commands are important but would only be passed on to the OTRSP box
for physical switching of headphones ?

ny4i, 2022-04-26:
Howie,

That is the right track. I am thinking I need something such as when we SWAPRADIOS, I send a command to OTRSP to swap the RX and TX.

There is also the question if there are manual controls on the OTRSP box that should change states within TR4W. From looking at the YCCC box, there are switches so I suspect if a switch is pressed, I will get a message from the control program and have to take the correct action.

Maybe we just need a state table.

Tom

ny4i, 2022-04-26:
One additional observation. Adding support for a brand new keyed may not be trivial just because of how much code has exceptions such as If WinKeyer...else ... It would be better to just put the CW sending code in one class and leave the decisions to each object based on keyer type and simply call functions. It's good to explore this but these changes would be after I add the Flex TCP/IP code and a few other changes.

n4af, 2022-10-11:
> Howie,
>
> That is the right track. I am thinking I need something such as when we
> SWAPRADIOS, I send a command to OTRSP to swap the RX and TX.
rr/ either TX1, TX2, or NIL
also commands r1,r2,r1s, and r2s

>
> There is also the question if there are manual controls on the OTRSP box
> that should change states within TR4W. From looking at the YCCC box, there
> are switches so I suspect if a switch is pressed, I will get a message from
> the control program and have to take the correct action.
Well, I am thinking the OTRSP box will do most of that standalone. Eg, it
has key line, headphones, and ptt lines that the switches will control...

n4af, 2022-10-11:
A lot of the commands can be skipped for now.
REQUIRED:
  The *TX1, TX2, RX1, RX2*, *RX1S, RX2S*, RX1R, and RX2R commands. RX1R and
RX2R can be treated as RX1S and RX2S. RTS as a PTT control if appropriate.
I am guessing Mode would be nice, but suspect TR4W can already control that
based on radio physical settings ?

ny4i, 2023-03-06:
I have a YCCC box in the mail on its way to me. I found a used one so I can implement and test it.

n4af, 2023-03-06:
WOW, that would be great Tom. I think Na8V dropped tr cuz no otrsp support.

Thanks, Howie

ny4i, 2023-08-22:
OK. I think this is next. It is not clear yet to me if I should use a socket connection or a serial connection. My YCCC box states the software listens on a socket. But it appears one can open the serial port via USB and talk to it directly.

ny4i, 2024-02-08:
Almost 10 years in the making...

I started on this patch for those keeping score.