TR4W/TR4W #65: An error in the cluster commands file causes an access violation uppon startup

ny4i, 2015-12-13:
I edited the cluster commands file and had an error. This caused the program to receive an access violation upon startup. It is possible this was due to a very long entry but this needs to be checked. At a minimum, it should handle and report the error rather than crash.

ny4i, 2015-12-14:
I was curious about this. One of the perils in Windows with the CopyMemory command is there is no bounds checking. You need to do this yourself. 

In EnumerateLinesInFile in src\TF.pas, exists the following code:

while FilePos < FileSize do
  begin
    if (MapBase[FilePos] in [#13, #10]) then
    begin

      if not NewLine then
      begin
        LastLine:

        LineSize := FilePos - StartPos;
        if LineSize > 0 then
        begin
          Windows.ZeroMemory(@TempString, SizeOf(TempString));
          TempString[0] := CHR(LineSize);
          Windows.CopyMemory(@TempString[1], @MapBase[StartPos], LineSize);
          if UpperCase then strU(TempString);
          Func(@TempString);
        end;
      end;

I created a very large filter statement and it exceeded the size of a shortstring. If this code truly wants to limit the size of the filter, then it should only read up to the length of a ShortString from the file. If that route is taken, then a message that the particular line exceeds the size limit would be useful. Another option would be to increase the size OR allow a continuation character in the file (like a ;) so we could read the entire line up to the ;. That would make for a better parser but at the very least, add the length limit so we never try to CopyMemory with  more that size(ShortString).

Note the error actually happens on the Func call (which is a passed-by-reference function call to be called depending upon the particular file being enumerated). This is a general way to enumerate files. (That's just for reference...)

n4af, 2016-01-12:
Tom -

Could you re-test ? 
This could be picked up by 4.44 --
4.43.9 Program check when write long line to telnet list box. Now limit < 120 char (NY4I)

ny4i, 2022-02-21:
This still fails. We could simply make the commands a TStringList and handle them manually with all this string manipulation.

ny4i, 2022-02-21:
One other thougfht...should we allow macro characters in this file.

So set/dx/filter call=N4AF or (not skimmer and spottercont=NA) or ((skimmer and unique>2) becomes
set/dx/filter call={\} or (not skimmer and spottercont=NA) or ((skimmer and unique>2)

with the {} indicating a Macro and the \ expanding to MyCall like in a CW message?

n4af, 2022-02-21:
Sounds interesting...