TR4W/TR4W #66: Send spot command (`), puts last call worked but current frequency

ny4i, 2015-12-13:
This is more of a question but I have noticed that if I work a station (S&P), then tune around a bit, if I then I forgot to spot the station, when I hit the spot key, the call is pre-filled, but the current rig frequency is pre-filled. if we pre-fill the call, shouldn't we pre-fill the frequency where we worked the station?

n4af, 2015-12-14:
I think this may defeat KISS. What if I work a station then qsy and hear a good mult but do not need him. Spot would put out the last station I worked. 

If performance is not degraded, perhaps we need a 'spot all' option ???
This would invoke menu_ctrl_sendspot on enter of every qso ...

73, Howie

ny4i, 2015-12-18:
Howie - I'm not quite following...The callsign is pre-filled with the last call I worked. I may not understand your comment about "What if I work a station then qsy and hear a good mult but do not need him. Spot would put out the last station I worked. ". That is exactly how it would behave today. The callsign will be pre-filled with your last station but the frequency you are on. The question is if ` is a shortcut for spot the last station worked? If so, then as it pre-fills this call today, I submit it should pre-fill the frequency. 

In your case of hearing a multiplier you do not need, did you still want to spot it for others? In that case, the spot call field needs to be updated. 

One case we could do is that if the current VFO does not match the frequency of the QSO, then leave the callsign field blank, but keep the current freq pre-filled. If the VFO does match the freq of the last station worked, then pre-fill both. That would defeat my idea of pre-fill wth the last QSO info.

Maybe the thing we need is a differentiation to spot last contact (including its call and freq) and a general Send Spot that is not pre-filled except for the call.

To me it comes down to the possible use cases plus these are only pre-filled. In any case, using the current VFO and last call logged is inconsistent so I think we should go one direction or the other as current freq and last call does not seem correct.

n4af, 2015-12-18:
On Fri, Dec 18, 2015 at 1:09 AM, Tom Schaefer <notifications@github.com>
wrote:

>
HI Tom -

Seems to me we offer the user:
1) no spotting (default)
2) Spot All
3) Spot current freq, call, contest (your suggestion)

Can you investigate adding sleep (20) to addspot ?
function TDXSpotsList.AddSpot label "ADD:"

I think we are in an impossible position- VE5ZX complains about autocq when
he runs spotting full out. I am not convinced inserting sleep is going to
be a cure- it will help autocq but not sure what the effect on spot flow
would be.--Maybe we should turn on the sleep and see what happens in a
large contest ?
To me, it should be user responsibility to filter spots in a large contest.

73, Howie

http://n4af.net

ny4i, 2022-02-19:
Spot last contact would be the case where we pre-fill the call, mode and frequency of the station we just worked. Maybe we add an option on the dialog to spot current frequency and mode as an override.