TR4W/TR4W #67: Need better error handling when cannot connect to cluster

ny4i, 2015-12-13:
If we cannot connect to the cluster for some reason, the error messages are not very good. Sometimes we get a Windows WinSock error or just Command Completed Successfully. This should be beef'ed up and return better, more descriptive errors. It might not hurt to check the server or IP via a ping to test connectivity to the cluster

closed 2020-07-02 by n4af

ny4i, 2020-07-03:
> Closed #67 via #412.

Oh no :)

You need to back that out :)

I update it so it’s in the repo not because it’s ready. That would make sone people mad if that got out yet. 

Thanks

Tom

Principal Solutions Architect
Better Software Solutions, Inc. 
727-437-2771

n4af, 2020-07-03:
Ok. I reverted issue 412.

73, Howie

PR #412 (2020-07-02): Telnet connect not in list
The fixes #392 .  Now an item in the cluster config is added to the server drop-down list at run time but does not modify TRCLUSTER file. Also added better messages if the the cluster cannot connect so this also fixes #67 .
commit 2c4e6da3 (2020-07-02): Alow connecting to cluster not in file

This allows a user to connect to a cluster in the telnet config option even if the cluster is not in the TRCLUSTER file. This closes #392. Also closes #67 to give a better error message if we cannot connect.