TR4W/TR4W #73: Highlight nearest frequency in bandmap

ny4i, 2015-12-13:
I like the tabular layout of the TR4W bandmap. Other programs have the VFO layout in a vertical dial but I do not like that as much. One thing that woud be nice is to have a black outline around the enearest spot or at least when we turn over one. The bandmap seems to always display the lower end regardless of where one is tuned. I had spots for both CW and SSB in the ARRL-10 so I had a good size list of spots. I would like to keep the band-map centered around my VFO frequency. 

This sounds familiar so perhaps this is already an option.

gm0gav, 2016-02-16:
Tom,
the band-map centered around my VFO frequency should work, look at the BandMap Display LImit command. It starts to centre once you go past the value set in BandMap Display LImit.Use that value to match to the spot capacity of your particular bandmap window. It would be nice in future to calculate the capacity of your window dynamically, then adjust the BandMap Display LImit.

Gav

ny4i, 2016-02-20:
I had a chance to try this and I agree that this works as you describe. I agree it would be a nice feature to change the internal bandmap display limit as the window is resized. That is what threw me that this did not happen automatically when I resized the window.