TR4W/TR4W #74: Add support to send logged QSOs to DXKeeper in real-time

ny4i, 2015-12-19:
DXKeeper opens a TCP/IP socket on port 52000. If we send the following to that socket, the contact is logged...

<command:3>Log<parameters:130><CALL:4>W1AW <RST_SENT:3>599 <RST_RCVD:3>599 <FREQ:7>7.13537 <BAND:3>40M <MODE:4>RTTY <QSO_DATE:8>20151219 <TIME_ON:6>064900 <EOR>

This would be for only casual operators that want to keep SpotCollector in sync for DX Chasing. This should be an option but I would not suggest enabling it for a serious contest effort.

ny4i, 2016-01-24:
I started working on this but it is more generic to send data to the same port that N1MM broadcasts data on. This way, any program that can read the data from N1MM's UDP broadcast can read this data. There so happens to be a bridge that listens to the N1MM UDP port and then sends a message to DXKeeper to log the message. This is how I did it.

It was surprisingly easy to get the fundamentals working. It took me about 2 hours and that was compiling the IndyProject controls. I used Indy in a past life for SMTP. The project is a set of classes that handles all kinds of networking protocols (including SSL for the tickets to send to getscores via https). I am beyond creating sockets by hand in Windows—that is not a productive use of anyone's time. Indy does a great job of this so why not use it. I will post more details on how I did it and how to add Indy to your individual Delphi library. indyproject.org has more info if anyone is interested.

I send just a hard coded QSO record in LogContact. Now I have to build the XML string from the actual QSL data in TR4W. I will also add some options to set the UDP port number and the IP of the computer to which to broadcast (defaults to 127.0.0.1).

n4af, 2016-01-24:
Ok, sounds promising Tom.

73, Howie

ny4i, 2016-01-26:
I changed the title of this as after doing some more research, there is an entire ecosystem out there that reacts to the N1MM UDP Broadcasts. I see rotor interfaces and SteppIR controllers that react to these UDP Broadcasts. I even see now that the Palstar HF-AUTO auto tuner will start adjusting its settings based on the Radio broadcast. I currently have the QSO broadcast done. I have to fill in some more information about the QSO to log and I also want to add both the ROTOR broadcast and the RADIo Broadcast. More on those formats in a subsequent post.

ny4i, 2016-01-26:
Here is the URL to the way N1MM sends Broadcasts: http://n1mm.hamdocs.com/tiki-index.php?page=UDP+Broadcasts

As many other programs are using this standard, I am implementing the same thing (I love standards and do not believe in Not-Invented-Here).

New Config Commands:

UDP_BroadcastAppInfo (Default False)
UDP_BroadcastContact (Default False)
UDP_BroadcastRadio (Default False)
UDP_BroadcastRotor (Default False)
UDP_DestinationPort (Default 12060)
UDP_DestinationIP (Default 127.0.0.1)
UDP_DestinationPortRotor (Default 12040)

Note that the Radio Broadcast is whenever the radio changes (or every 10 seconds). 

Here is the data sent for the Radio broadcast:

<?xml version="1.0"?>
   <RadioInfo>
      <StationName>CW</StationName>    *
      <RadioNr>2</RadioNr>    **
      <Freq>1400100</Freq>
      <TXFreq>1400200</TXFreq>
      <Mode>CW</Mode>
      <OpCall>NY4I</OpCall>
      <IsRunning>True</IsRunning>
      <FocusEntry>hWnd</FocusEntry>
      <Antenna>2</Antenna>
      <Rotors>GreenHeron</Rotors>
      <FocusRadioNr>2</FocusRadioNr>
   </RadioInfo>

Here is the rotor info:

<N1MMRotor>
     <rotor>Green Heron</rotor>
     <goazi>55.0</goazi>
     <offset>0.0</offset>
     <bidirectional>0</bidirectional>
     <freqband>14</freqband>   *
</N1MMRotor>

Stop Rotor
<N1MMRotor>
      <stop>
            <rotor>Green Heron</rotor>
            <freqband>21.0</freqband>
      </stop>
</N1MMRotor>

ny4i, 2016-01-26:
Here is a sampling of the programs I found that use the broadcasted data from UDP. The benefit to TR4W is that if we send the same broadcast, this program will now work with these things:

*[Qsorder](http://www.hamradiomap.com/qsorder/) - Contest Audio Recorder 
*[ClubPI](http://www.nx1p.net/clubpi/what-is-clubpi/) - DXPedition/Contest aggregator to send contacts to ClubLog in real-time 
*[PstRotator](http://www.qsl.net/yo3dmu/index_Page346.htm) - Rotator Control Software 
*[DXKeeper Bridge](http://www.dxlabsuite.com/N1MM-DXKeeper/index.htm) - Allows contacts to be sent to DXKeeper in real-time 

Please note that for other logging programs (HRD, etc), using this broadcast mechanism means we do not need to support each and every logger. If a developer wants to add contest QSOs to the log in real-time, that program either needs to listen for the broadcasts or someone needs to write a bridge to accept the UDP broadcast and send to whatever program over an offered interface (DDE, TCP, etc). This puts the burden on a different development team (or one one of us if there is a logger we want to implement—I use DXKeeper so it is done for me). As the same broadcast mechanism will work for N1MM and TR4W, they get more bang for the buck and would see more use than just the TR4W user base.

ny4i, 2016-02-01:
Branch 4.44.9 contains this code. The commands test (English) has been updated. Note that I did not include the AppInfo broadcast as that was very N1MM specific and in fact, it does not appear to broadcast that either.

closed 2016-02-01 by ny4i

n4af, 2016-02-01:
Hi Tom -

I suspect this d/n work for LOTW, but you implied it does for ClubLog.
What would be needed to upload to ClubLog real time ??

Tks, Howie

http://n4af.net

n4af, 2016-02-01:
Also, what would be involved in real-time GetScores from this ?

73, Howie

http://n4af.net

ny4i, 2016-02-01:
Release Note
In order to support the growing ecosystem of software that makes use of the N1MM UDP Broadcast data, TR4W now supports the same formatted UDP broadcast data for Contacts, Radio updates and rotator updates. In order to use this, you have to enable the UDP Broadcast Options in the configuration interface (CTRL-J). By default, the broadcast is disabled, but if you enable a particular one (UDP Broadcast Contact for example), then whenever a contact is completed, TR4W will broadcast data on the default UDP port 12060 (Note: The port number can be changed in the config options as well).

Typical use cases are any software that makes use of this data. For example, if you would like to do QSO recording using the qsorder software, because qsorder makes use of the broadcasts from N1MM, it can also use the broadcasted data from TR4W. 

Another option is to send contacts to ClubLog in real-time. For this, you have two options. First, is a system called ClubPi (written by ). ClubPi runs on a Raspberry Pi and will send contacts in real-time or whenever it has a network connection. This was created to support DXpedition uploading of contacts from the logging software. Another easier option is a Windows program called N1MM-ClubLog Gateway. More info is here: http://www.n2amg.com/software/n1mm-clublog-gateway/

In short, any programs that makes use of N1MM broadcast data can now be used with TR4W. They say that imitation is the kindest form of flattery, so I greatly appreciate the trail that the N1MM team blazed and the other developers that have implemented the many projects that use this data.

ny4i, 2016-02-02:
After testing with ClubPi (which uses a Raspberry Pi to connect to CLubLog for DXPedition support), I found some missing fields. This now works properly. I am able to broadcast to a separate Pi running on my network as well as to the local system to the DXKeeper gateway. Cool stuff. 

I updated the 4.44.9 branch but I noticed that Howie updated my changes into Master. I hope that is OK to pull the new changes from 4.44.9.

ny4i, 2016-02-02:
Just to reference, this is now the command that the program sends to the UDP port...

<?xml version="1.0"?>
<contactinfo>
	<contestname>GENERAL QSO</contestname>
	<contestnr>3</contestnr>
	<timestamp>2016-02-02 00:09:29</timestamp>
	<mycall>NY4I</mycall>
	<band>24</band>
	<rxfreq>2489407</rxfreq>
	<txfreq>2489620</txfreq>
	<operator>NY4I</operator>
	<mode>CW</mode>
	<call>LY7Y</call>
	<countryprefix>LY</countryprefix>
	<wpxprefix>LY7</wpxprefix>
	<stationprefix>LY7</stationprefix>
	<continent>EU</continent>
	<snt>599</snt>
	<sntnr>29</sntnr>
	<rcv>599</rcv>
	<rcvnr>0</rcvnr>
	<gridsquare></gridsquare>
	<exchange1> </exchange1>
	<section></section>
	<comment></comment>
	<qth></qth>
	<name></name>
	<power></power>
	<misctext></misctext>
	<zone>29</zone>
	<prec></prec>
	<ck>0</ck>
	<ismultiplier1>0</ismultiplier1>
	<ismultiplier2>0</ismultiplier2>
	<ismultiplier3>0</ismultiplier3>
	<points>1</points>
	<radionr>1</radionr>
	<RoverLocation></RoverLocation>
	<RadioInterfaced>1</RadioInterfaced>
	<NetworkedCompNr>0</NetworkedCompNr>
	<IsOriginal>True</IsOriginal>
	<NetBiosName>T</NetBiosName>
	<IsRunQSO>1</IsRunQSO>
</contactinfo>