TR4W/TR4W #91: Multipliers Remaining Display

ny4i, 2016-02-08:
I saw an interesting post on the N1MM mailing list referring to a multipliers left display where you remove multipliers until none are left (meaning you worked them all). The reason I am adding this is the poster referenced how TRLog used to do it and it reminded me that TRLog indeed used to do that. I wonder if we should provide an alternate display to do that again. Here is the post:

> Bob, W9ZV said, I too am thumbs up for a "multipliers left" or subtractive model for the section display of mults. 
> 
> not advocating the removal of what mult display you have there now, but add an alternate display/option for sections? 
> 
> remember the way TRLOG used to work for this ?? the list just got smaller and smaller (certainly not nearly as wide) as you burned off another section, until the window (screen area) was clear, and then its Miller Time as you reach to fill up your Clean Sweep mug ! lends itself to being left up all the time if you like. 
> 
> no need to do mental interpretations of little boxes/band etc... nor a need to scan down the columns with your MKII eye balls in focus. In a subtractive model, there is absolutely no reason to display sections already worked. Furthermore, the dialog box needed can be much smaller to save on the most precious asset we have on the confuser - screen space. much easier to remind yourself what you need to go after. particularly for us little pistols... 

> KISS principle... Keep the Atomic Powered Swiss Army Knife display for others who want it. 
> 
> of course this display is most useful for things like SS... countries wouldn't apply (too many will always be needed) etc. 
> 
> I had put a request for this nearly two years ago, kinda fell on the floor since focus I think was on MM+ 
> 
> maybe re-open this crypt again? . 
> 
> -bob, w9zv

gm0gav, 2016-02-08:
Hi Tom,

its already there, at least for domestic mults where you have an 
absolute multiplier list. Try the latest version with UBA contest, then 
windows / remaining mults.

A few issues with the new version experienced this weekend, strange 
issue where some bandmap spots were getting the wrong band allocated (or 
so it looked like). I need eed to study it further.

73 Gav

n4af, 2016-02-08:
Hi Gavin -

I noticed in marathon that many of the qsos showed in call window as 'mult'
because of new member number (interpreted as mult ?).
Not a real problem as they logged correctly.

73, Howie

http://n4af.net

gm0gav, 2016-02-08:
I noticed that also Howie, It is most likely that the Marathon contest 
definition is not quite right and needs to be tweaked. By that I mean it 
is treating the foc number as a multiplier or expecting a dom file with 
them in, when its just the callsign that counts.

73 Gav

n4af, 2016-02-08:
Ah, yes. Makes sense.

73, Howie

http://n4af.net

gm0gav, 2016-02-16:
I think this can be closed? Support there and appears to work as designed

ny4i, 2016-02-26:
I tested with ARRL Field Day. The mults I worked are highlighted, but they are still there. I believe this request was that the mults disappear totally from the list.