TR4W/TR4W #92: Age and QSO number written as one word

n4af, 2016-02-08:
Suddenly, there was a problem since 4.44.9 and above.
If      EXCHANGE RECEIVED = AGE AND QSO NUMBER
accidentally age and number written as one word (24001),
then there is a program reset. See the clipboard 10, then 11.
When separate records (24 001) - all OK.
4.44.8 and below are not the problem.
(de UR7QM)

gm0gav, 2016-02-08:
Can we find out the contest where this happens?

n4af, 2016-02-08:
This is a contest based on modifcation of General QSO.

73, Howie

CONTEST = GENERAL QSO
CONTEST NAME = R0J-CONTEST
AUTO DUPE ENABLE CQ = TRUE
AUTO DUPE ENABLE S AND P = TRUE
BAND = 80
DIGITAL MODE ENABLE = FALSE
DOMESTIC MULTIPLIER = NONE
DUPE SHEET AUTO RESET = TRUE
DX MULTIPLIER = NONE
EXCHANGE MEMORY ENABLE = TRUE
EXCHANGE RECEIVED = AGE AND QSO NUMBER

http://n4af.net

n4af, 2016-02-08:
Fixed in 4.45.2
Validation now  requires a minimum of two fields entered when
EXCHANGE RECEIVED = AGE AND QSO NUMBER

closed 2016-02-08 by n4af

ny4i, 2016-02-08:
Howie, I already had a 4.45.2. Did you just add this fix to that branch?

Tom
Principal Solutions Architect, ITIL®v3 
Better Software Solutions, Inc.
727-437-BSS1 (727-437-2771)
@BSSI_Consulting

n4af, 2016-02-08:
no. it was changed in issues to 4.45.7 - which I have not pushed up yet...

http://n4af.net

ny4i, 2016-02-08:
You had fixed in 4.45.2 which threw me.

As it turns out, I found a potentially critical error in the UDP code that could cause an access violation if a configured serial port was not available.

If you have not merged 4.45.2 yet, I can make the fix there. It’s a one lines but I want to test it properly.

Tom
Principal Solutions Architect, ITIL®v3 
Better Software Solutions, Inc.
727-437-BSS1 (727-437-2771)
@BSSI_Consulting

> On Feb 8, 2016, at 2:18 PM, Howie Hoyt <notifications@github.com> wrote:
> 
>  Fixed in 4.45.2

n4af, 2016-02-08:
Hi Tom -

AFAIK, I fixed at 4.45.7
I just pushed a comment correction to indicate 4.45.7 in the logstuff
comment.

73, Howie

http://n4af.net