TR4W/TR4W #93: System faults running RBN and TR4W 4,45

n4af, 2016-02-08:
Access violation running with RBN this weekend.
Backing off to filtered spots (VE7CC) circumvented the problem.

Faulting application name: tr4w.exe, version: 0.0.0.0, time stamp: 0x2a425e19
Faulting module name: tr4w.exe, version: 0.0.0.0, time stamp: 0x2a425e19
Exception code: 0xc0000005
Fault offset: 0x000988af
Faulting process id: 0x242c
Faulting application start time: 0x01d160ee8729cfcb
Faulting application path: C:\Program Files (x86)\TR4W\tr4w.exe
Faulting module path: C:\Program Files (x86)\TR4W\tr4w.exe
Report Id: 26adf291-dc33-4163-895c-a11b3f8c9187
Faulting package full name:

ny4i, 2016-02-08:
Howie - 

Are you and Gav connecting to the same RBN server and port? Can you share what server and port you use. Also, are there any filters at all?

How quick can you make it fail? I ask as I wonder if I could connect as your call and see if I can make it fail as fast.

n4af, 2016-02-08:
Hi Tom -

No, I just use telnet.reversebeacon.net:7000

Whilst it is rolling in, log dummy qsos.

If I call correctly, I had operated for some time before the problem
occurred but then it was frequent.

CQWWRTTY is coming up, but I am woefully ignorant of the volume of spots
generated by rtty.

Question:
any thoughts re: this comment from k6xx ?
(we have a problem where separate problems are not opened and it post to
tr4w forum comes in under wrong heading)
"
Anyone else having CW Keying problems via the parallel port with 4.45?
Keying is now choppy (occasional dashes have "holes" in them), and after
sending CQ via the ENTER key successfully several times, it instead sends
"FA" (or similar). Other messages are also corrupted sporadically.

I reverted to a September release and all is well.

73 de Bob, K6XX
"
73, Howie

http://n4af.net

ny4i, 2016-02-08:
I saw that. I thought I asked a follow up question on the reflector but I could be wrong.

Do you have any gear to test with the parallel port anymore?

Tom
Principal Solutions Architect, ITIL®v3 
Better Software Solutions, Inc.
727-437-BSS1 (727-437-2771)
@BSSI_Consulting

n4af, 2016-02-08:
unfortunately, no.

73, Howie

http://n4af.net

gm0gav, 2016-02-08:
I can test parallel port, but not until next weekend as away on business. 

73 Gav

n4af, 2016-02-08:
That would be great.

Thanks, Howie

http://n4af.net

gm0gav, 2016-02-13:
Guys,

I have got up to date with 4.45.5 locally. I set up to use parallel port 
cw & PTT, rather than winkey. Played around for 30 minutes on both 
radios and the CW was perfect, no drops or funnies.

So either it's been fixed or I cant reproduce K6XX's issue

73 Gav

n4af, 2016-02-13:
Thanks Gavin -

I had to send one of my rigs in for repair.
But going to crank up rbn and make some dummy entries to test 4.45.6

73, Howie

http://n4af.net

n4af, 2016-02-25:
closing - not able to reproduce

closed 2016-02-25 by n4af