TR4W/TR4W #94: CWbyCAT does not fully support SO2R

n4af, 2016-02-08:
CWbyCAT does basic switching but needs to kill active on switch to inactive.
also cq on inactive rig may not function.

ny4i, 2016-02-08:
I did testing with this with my two radios, but I do not recall doing CQ on the inactive radio. I was CQing on the primary radio and working S&P on the inactive one.

Thanks for finding this…

Tom
Principal Solutions Architect, ITIL®v3 
Better Software Solutions, Inc.
727-437-BSS1 (727-437-2771)
@BSSI_Consulting

n4af, 2016-02-08:
Hi Tom -

Well, I think it puts the operator into Multi category.

We need to follow WK convention and dump the CW into a buffer that can be
flushed.
Would be cool to use the WK buffer routine but not sure how practical that
is.

73, Howie

http://n4af.net

ny4i, 2016-02-08:
Oh so you are saying it was sending CW on two rigs at the same time?

Tom
Principal Solutions Architect, ITIL®v3 
Better Software Solutions, Inc.
727-437-BSS1 (727-437-2771)
@BSSI_Consulting

n4af, 2016-02-08:
I think what happens is there is no gating on the CW.
If I recall correctly, whatever is being sent on rig1 goes to completion
even is sending is initiated on rig2.

73, Howie

http://n4af.net

n4af, 2016-02-10:
I have a one line change to SWAPRADIOS that enables kill of active radio and switch to inactive radio. 
It allows for immediate kill of the cq and swap radios over to the s&p. It also works with alt-d.
This should ensure SO2R ops are not reclassified MULTI because the present code continues sending the message on active rig WHILST ALSO sending on the inactive rig.

What this does not do is CQInactive. That is probably likewise simple, but that could be done later.

The fix is currently in the beta directory:
http://tr4w.net/beta/tr4w_setup_4_45.cwbycatso2r.exe

ny4i, 2016-02-15:
I added some code to fix this but it is NOT complete. 
*** DO NOT ADD THIS TO MASTER YET ***
The Switch boolean is hanging me up when I am in S&P mode and I try to send Call after <01> <02> We are switching back to the activeradio versus SwapRadio.

n4af, 2016-02-15:
Hi Tom -

I am trying to understand the nature of the problem.
RU saying when in S&P and invoking <01><02> ?

If so, I would agree it deviates from Wkey-- but, technically, the command
should be
<01><02>*<03>CQMODE<04>*CQ \

Alternatively, just run a CASE of TEMPCHAR after LogSubs1 has checked for
controlA thru controlE where he sets
SetOpMode(CQOpMode); ?

Or, perhaps I misunderstand the problem ?

73, Howie

http://n4af.net

ny4i, 2016-02-15:
Her is the scenario:

In S&P mode with K3 as ACTIVE radio and 9100 as INACTIVE radio.

I have F3 programmed with <01><02>CQ \\

I hit F3, radio2 (9100) sends CQ. I type a callsign it goes out on K3 (this is incorrect I believe).

Note that if I am in CQ mode, when I hit F3 Radio2 sends out the CQ. I type a callsign and radio2 now becomes ctive and radio2 sends my callsign and my exchange.

But, I believe the use case is that we want to send CQ on inactive radio when we are in S&P mode. Is this correct?

If so, I cannot figure out how it works differently except there is some code called CheckInactiveRadioCQ (or something like that).

Tom

Principal Solutions Architect, ITIL®v3 
Better Software Solutions, Inc.
727-437-BSS1 (727-437-2771)
@BSSI_Consulting

ny4i, 2016-02-16:
After discussion with Howie, if you want to send CQ on second radio from S&P mode, then the command CQMODE has to be used in the macro. <01><02>*<03>CQMODE<04>*CQ \\ \\

closed 2016-02-16 by ny4i