TR4W/TR4W #95: Issue with CWByCAT and AUTOSEND/AUTO Terminate

ny4i, 2016-02-09:
Reported by Marv, N5AW...

In CWBYCAT and when using AUTO SEND, strange behavior was experienced where when the contents of the call window are sent, the cursor was not going to the exchange window.

There is also a case when hitting ESC to stop the sending that the program can end up sending anything updated in the call window. It appears that the something is not getting reset to the same state before the auto send count was reached.

As Howie said, AutoSend is the gift that keeps on giving :)

Here is the text of Marv's report...

> We are in Colorado with limited Internet access. I encountered some fairly serious problems with 4.45.0 when using it for the Marathon last weekend and confirmed they still exist with 4.45.7 this morning. The most serious is something has really screwed up the Winkeyer interface. Some of the problems are the same I've had in the past (but not for a while) that could be temporarily fixed with the Winkeyer reset (but not with this version) - Esc not working to stop sending, immediately jumping to the exchange box after sending the letters specified in Auto Send Character Count when using Auto Call Terminate. The worst though is it would randomly start sending one of the exchanges repeatedly with no way to stop it short of closing the program down. I switched to the CAT keying option for the K3, turned off the Winkeyer and those problems went away. However there are two problems with this option. When you enter a call with Auto Call Terminate it sends the call but then you have to hit enter for it to send the exchange. Also if you enter a call when in Auto Call Terminate mode and then hit Esc it activates keyboard sending (F10). You have to hit F10 to get out of it. There may be some other keyboard sequences that get you into F10 mode too - not sure. It happened several times to me over the weekend.
> 
> I really like the CAT keying and paddle sync options but hope those  bugs can be fixed. The Winkeyer problems are going to be bad news for anyone using a Winkeyer! The Winkeyer problem is not there in 4.44.0 so perhaps it has something to do with the CAT keying addition.
> 
> Marv N5AW

n4af, 2016-02-09:
I have a fix for the escape/autosend/f10 problem.
Will try to build it this afternoon.

73, Howie

http://n4af.net

n4af, 2016-02-09:
Moved above escape during autosend issue to ISSUE # 96
This issue if for Entering a call with autosend requires enter key, using CWByCAT.

n4af, 2016-02-09:
Fixed 4.45.9
add Process Return to  CallWindowKeyDownProc of MainUnit

closed 2016-02-09 by n4af