TR4W/TR4W #100: CWBYCAT: Hitting ESC does not stop Icom radio from sending code

ny4i, 2016-02-11:
When one hits ESC, the icom radio should stop sending the string it is working on. Add code to send a step sending CW message if possible.

n4af, 2016-02-11:
Hi Tom -

Can you verify MainUnit invokes Escape_proc and the InitializeQSO is then
invoked ?

Tks, Howie

http://n4af.net

ny4i, 2016-02-11:
You mean verify the change you made? Sure I can. As it works just fine for the K3, I suspect I omitted the message to the Icom to take it out of xmit mode.

Tom
Principal Solutions Architect, ITIL®v3 
Better Software Solutions, Inc.
727-437-BSS1 (727-437-2771)
@BSSI_Consulting

n4af, 2016-02-11:
rr
73, h

http://n4af.net

ny4i, 2016-02-12:
Howie, Please merge this into Master. it fixes the code so hitting ESC on an Icom radio stops sending CW. -- Thanks. Off to finish packing for Orlando.

n4af, 2016-02-12:
Hi Tom -

I do not know how to proceed.
It appears you put two branches into the remote ???
It also appears FieldDayChanges may have gone up without pulling down the
latest branch (4.45.3) ??

I had to restore my last backup to recover the local repo, but lost my
local changes.

I suppose I will explore removing remote branch FieldDayChanges but am
having a tough time trying to sort this out and move forward from 4.45.3

73, Howie

http://n4af.net

ny4i, 2016-02-12:
Howie,

Did you not get the part of my email where I said to NOT do anything with FieldDayChanges? We would wait until the dust settles. That is just there for safekeeping?

The only change you needed to incorporate was the CWBYCAT_icom_notstopping.

I always do a fresh pull from origin/master and merge it into my current branch, clean up any conflicts, the re-commit. So I always have the latest changes in master in my branch before I push the changes to origin. After that, my client says they are up to date.

Thanks,

Tom
Principal Solutions Architect, ITIL®v3 
Better Software Solutions, Inc.
727-437-BSS1 (727-437-2771)
@BSSI_Consulting

n4af, 2016-02-12:
Ok, I am not sure I have integrity in my local master.
Normally, I do a 'pull origin' but it pulls in any new branches.

Could you check 4.45.4 and see if it is ok ?

73, Howie

http://n4af.net

ny4i, 2016-02-12:
Howie,

It looks like this has the Field Day code in it too. If you can wait until Sunday evening, I can make sure it is OK before you go further. Are you sure you do not want to rollback to before you merged these two, get the fresh copy from origin and go from there? 

We have to discuss your workflow in Git as you seem to have some issues :) I wonder if you are doing a git fetch before you do a git pull in master?

The Field Day code is fully tested and ready. I just did not want to complicate matters by adding more features before ARRL CW but it can work just fine and no one will end up hitting the code anyway.

Tom
Principal Solutions Architect, ITIL®v3 
Better Software Solutions, Inc.
727-437-BSS1 (727-437-2771)
@BSSI_Consulting

n4af, 2016-02-12:
ok, i rebuilt my local and remote.

73, Howie

http://n4af.net

ny4i, 2016-02-16:
This code was fixed by adding code to the CWBYCAT to stop sending. This was also fixed on the second radio.

commit 8fc96408 (2016-02-12): The code for Icom in StopSendingCW was not implemented. I implemented this by sending a CW string of Chr(255) 'FF'. This is from the Icom manual. Note that this SendCW (17h) CI-V command does not have a subcode.
This patch fixes #100.
closed 2016-02-16 by ny4i