TR4W/TR4W #101: Pressing escape removes call

n4af, 2016-02-12:
Pressing escape should just stop sending, not remove the call

n4af, 2016-02-12:
Closed 4.45.5

ny4i, 2016-02-13:
Did I introduce that regression?

Principal Solutions Architect
Better Software Solutions, Inc.
727-437-2771

Closed #101 <https://github.com/n4af/TR4W/issues/101>.

n4af, 2016-02-13:
No. N5AW complained that cwbycat was not working on esc.
I had it so escape stopped the cw but also cleared the call.
So changed it so esc only stops the cw. He then noted that hitting escape
twice, like wkey,
d/n clear the call.

Right now I told him to use alt-w. But i would like you to review the code
to ensure I d/n stomp on what you had added.

73, Howie

http://n4af.net

ny4i, 2016-02-20:
This is not working right. Here is the scenario with 4.46.4:

1. In S&P mode, type a call and hit enter.
2. Program sends MYCALL and waits in the exchange box for input.
3. Hot ESC. no noticeable action. 
4. Hit ESC again and program clears the call window and goes to CQMODE.

This was working so I think the ESCAPE_proc is being too aggressive.

ny4i, 2016-02-26:
After the 4.46.7 merge, this is still a bug. 
Howie - You are familiar with the ESC code, can you take a look at this please? This was in the ARRl DX CW test.

n4af, 2016-02-26:
I will try to get to this as next bug fix.
99% chance it is that I am not resetting the switch set to do the clear.
Works for CQMode but not SAPmode.

http://n4af.net

n4af, 2016-03-01:
fixed 4.46.7

closed 2016-03-01 by n4af