TR4W/TR4W #104: Support for OmniRig

gm0gav, 2016-02-16:
Add the option to radio to use Omnirig. This allows serial ports to be shared with other programs like skimmer. It also removes the need TR4W to natively support every radio.

ny4i, 2016-02-28:
FAQ: http://www.dxatlas.com/omnirig/OmniRigFAQ.txt

Website for OmniRig: http://www.dxatlas.com/omnirig/

n4af, 2016-02-29:
Hi Tom -

One thing I neglected to mention.
I circumvented the ts590 model problem by declaring the model a 'K2' and
changing the buffer appropriatedly.
Not good for general use, but as a circumvention.

The results were different than now:

no buffer space problem
but pressing escape did not halt sending

Now the above is reversed

73, Howie

http://n4af.net

ny4i, 2021-02-03:
Another data point to support doing this is for users of HD-SDR. I had a request to allow using HD-SDR at the same time at TR4W for high control. Since Omnirig would control the radio, multiple clients could then talk to it including TR4W and HD-SDR.

ny4i, 2021-02-03:
One important point for when this is done (from their FAQ):

Q2: What are the terms of use?

A: You can use Omni-Rig in your projects for free. You are allowed to include
it in your setup package. The only requirement is that you mention in the
product documentation and on the web site that your program uses the Omni-Rig
technology, and include a link to the Omni-Rig web site.