TR4W/TR4W #105: Bandmap Band is changed to the Active radio when it calls CQ

gm0gav, 2016-02-16:
Bandmap Band is changed to follow the Active radio when it calls CQ.When tuning on inactive radio, bandmap band correctly follows inactive radio. It previously would only change bands * if * the active radio changed frequency ( or if the inactive radio changed band)

It now flips to the band of the radio calling CQ, every time CQ is sent. This works correctly in 4.37, the Active / Inactive radio logic appears to have changed since then.

ny4i, 2016-02-16:
How are you generating CW? WK, Serial or CWBYCAT?

Tom
Principal Solutions Architect, ITIL®v3 
Better Software Solutions, Inc.
727-437-BSS1 (727-437-2771)
@BSSI_Consulting

gm0gav, 2016-02-16:
It was LPT for testing purposes, will try with winkey.

ny4i, 2016-02-16:
Thanks. I was just curious if anything I did in the CWBYCAT area could have caused this but my changes all require CWBYCAT to be enabled.

Perhaps Howie knows if he did any work int his area with the bandmap.

Tom
Principal Solutions Architect, ITIL®v3 
Better Software Solutions, Inc.
727-437-BSS1 (727-437-2771)
@BSSI_Consulting

gm0gav, 2016-02-16:
This issue is present on winkey, LPT and CWBYCAT. It is ok in 4.43, appears to have been introduced in 4.44 or perhaps 4.45.

gm0gav, 2016-02-17:
I have now fixed this, not sure why it started happening but resolved by adding an extra filter to the code which selects   BandMapBand to only flip it when a radio has changed frequency. uRadioPolling line 2145 & 2169 add

    if ((rig.FilteredStatus.Freq <> BandMapCursorFrequency) and (rig.FilteredStatus.Freq <> rig.LastDisplayedFreq)  or                      //GAV BETA
      (BandMapMode <> ActiveMode)) and (rig.FilteredStatus.Freq <> 0) then
    begin
      SpotsList.DisplayCallsignOnThisFreq(rig.FilteredStatus.Freq);             
      BandMapCursorFrequency := rig.FilteredStatus.Freq;
      BandMapBand := ActiveBand;
      BandMapMode := ActiveMode;
      DisplayBandMap;
    end;

gm0gav, 2016-02-17:
We cant close this issue, the original problem also prevents TuneDupeCheck working correctly. So back to the drawing board!
 It seems like 
procedure ProcessFilteredStatus(rig: RadioPtr);
 is being called when ActiveRadioPtr is pointing to the incorrect  radio.

ny4i, 2016-02-19:
Gav - I "fixed" this by adding some checks if FList is Assigned. The issue was FList was nil for some reason. As I was talking to Howie, I had a thought...

If I delete a single spot, does that index become invalid? I wonder if it is possible that when deleting a single spot some index is not being updated and when we try to walk through the entire spot table for the given band, the code tries to remove a spot that is no longer there. 

I also had a thought that this might be more likely to happen now that the bandmap is not updated as often. I wonder if there is a timing hole where the spot could be deleted, yet the count is still the same so we try to delete it. That would have to do with concurrency and resource locking to ensure bandmap operations are atomic. 

I did testing and while calling CQ on the active radio, when I tune on the inactive radio, the band map switches. The bandmap stays on the inactive band until the radio stops sending and then it reverts back to the active band.

Is is supposed to stay on the inactiveband until we start a CQ again? If so, then I would guess something at the end of the CW is selecting the active band and this is causing the bandmap to switch.

Thanks!

ny4i, 2016-02-19:
One more item, I also noticed that when I was CQing on the active radio, when I clicked on a bandmap entry for the inactive radio, it switched the activeradio to that frequency (bad thing I presume). Is it supposed to work that while I am calling CQ, if I click on the inactive bandmap, then I Tune on the inactive radio?

Please note that I am new to the SO2R stuff so if there is some program option I need to set for this, please let me know.

n4af, 2016-02-19:
Hi Tom -

There is a right click option on bm entries re: qsy inactive radio

73, Howie

http://n4af.net

n4af, 2016-02-19:
integrated into 4.46.4

gm0gav, 2016-02-19:
Tom & Howie,
4.46.4 looks fine as far as faulting when you delete a spot and then delete all spots. All appears well there.

Tom you are correct,  
Switching bandmap band  when you change frequency = good, 
Switching bandmap band  when you key = bad, 

The switch should take place to follow the radio that last changed frequency, meaning that radio is tuning looking for stuff to work. If your CQing and working stuff, your not goint to be too interested in what is happening on your CQing / run band.

The easiest way to fix this is to change line 2079 in uRadio Polling to

 for TempInteger := 0 to SizeOf(RadioStatusRecord) - 2 do         

This prevents it checking TXOn boolean for a status change.

73 Gav

gm0gav, 2016-02-19:
I have tested with the change above in place, bandmap switching and AltDTune works properly with this in place

73 Gav
.

ny4i, 2016-02-19:
Gav,

Unless I am misunderstanding…I do need the TXOn to trigger a status change as that tells me when I am done sending CW (in CWBYCAT).

I see your subsequent email about it worked. I would like to test with CWBYCAT and hitting ESC while sending to ensure it stops.

Tom
Principal Solutions Architect, ITIL®v3 
Better Software Solutions, Inc.
727-437-BSS1 (727-437-2771)
@BSSI_Consulting

ny4i, 2016-02-19:
Is this change in a branch I can pull and test? Or is it in master? I will run this weekend wth this version.

Thanks,

Tom
Principal Solutions Architect, ITIL®v3 
Better Software Solutions, Inc.
727-437-BSS1 (727-437-2771)
@BSSI_Consulting

gm0gav, 2016-02-19:
Tom
CWBYCAT still works ok, Esc kills the message and get End on K3 display. The code after this change appears all to be to do with updating of freq display etc, rather than the business end of CAT.

I have not pushed this, better if you study & test the implications first. 

73 Gav

ny4i, 2016-02-19:
I understand. That was all I was concerned about regarding TX status.

If you make it a branch, I can pull that and test it out. You can also make it a pull request if you want to do it the “Git” way of having team members evaluate a branch for potential inclusion into master.

Thanks,

Tom

Principal Solutions Architect, ITIL®v3 
Better Software Solutions, Inc.
727-437-BSS1 (727-437-2771)
@BSSI_Consulting

ny4i, 2016-02-20:
I remembered. It was auto send that needed to tx status. It waits to start
the timer until sending is done ( indicated by tx off). This is in CWBYCAT.

Principal Solutions Architect
Better Software Solutions, Inc.
727-437-2771

gm0gav, 2016-02-20:
Both Auto CQ and Autosend appear to be working, both escaping correctly. 
Its looking good as far as I can see, will be on in ARRL for a run later 
so will see how that goes.

n4af, 2016-02-24:
The little I used these in ARRL, all seemed ok here.

gm0gav, 2016-02-25:
I played in ARRL and this is ok

n4af, 2016-03-07:
Tom - Can this be closed ?
Tks, h

gm0gav, 2016-03-08:
Partly resolved, TuneDupeCheck works in 4.47.3. But Bandmap still switches to CQing radio on every CQ

gm0gav, 2016-03-08:
Now resolved in  4.47.4. Bandmap band only changes as a result of a change in frequency.

closed 2016-03-08 by gm0gav