TR4W/TR4W #106: Clearing all spots causes an Access Violation

ny4i, 2016-02-17:
When right clicking on the spot window and clicking to remove all spots, the program faults with an access violation.

The place this stops is in uSport.pas at line 382
````delphi
function TDXSpotsList.Get(Index: integer): TSpotRecord;
begin
  //  if (Index < 0) or (Index >= FCount) then Exit; //ERROR(@SListIndexError, Index);
 ** Result := FList^[Index];**
end;
````

n4af, 2016-02-17:
Thanks for your detective work Tom.

I enabled the commented test, and that circumvents the problem.
I will shortly upload beta-1

Am sure Gavin is in Z land, but perhaps he can review tomorrow.

73, Howie

http://n4af.net

gm0gav, 2016-02-17:
This is a strange one, I have been trying to replicate on 4.46.beta and 
it was all fine until I deleted an individual spot, then did a remove 
all spots.

Will have a look tonight

73 Gav

n4af, 2016-02-24:
Gavin -

Can this be closed ? Tks, Howie

ny4i, 2016-02-24:
I saw a related issue. When I cleared the band map, the display changes to
strange characters momentarily. It's as if the band map is cleared but a
final display shows strange characters.

Principal Solutions Architect
Better Software Solutions, Inc.
727-437-2771

gm0gav, 2016-02-25:
I am sure this can be closed

ny4i, 2016-02-25:
I will turn this into another one then. The flash of garbage characters is a bug. I was thinking since it was related to clearing all that we could use this issue but it’s easy to open another one.

Thanks,

Tom

Principal Solutions Architect, ITIL®v3 
Better Software Solutions, Inc.
727-437-BSS1 (727-437-2771)
@BSSI_Consulting

closed 2016-03-07 by n4af