TR4W/TR4W #107: Add Translations for new constants

ny4i, 2016-02-17:
There are three new constants. This issue is to ensure they are updated in the language files.

Note these are related to ARRL Field Day so translation may not be urgent except for Spanish for Region 2.

The constants are:

- TC_IMPROPERTRANSMITTERCOUNT           = 'FD transmitters must be between 1 and 99.';
- TC_IMPROPERARRLFIELDDAYCLASS          = 'Field Day class must be A, B, C, D, E or F.';
- TC_ARRLFIELDDAYIMPROPERDXEXCHANGE     = 'DX Station exchange must be "DX".';

n4af, 2016-02-24:
I have added these constants to src\lang\tr4w_consts_eng.pas on my PC.
I do not see the call to the constants in the extant code so not sure I need to build this in yet.
It will be automatically picked up on the next build.
Thinking translation to other languages not required -- we do not have a translator for spanish, tho I suspect LU5DX could be 'volunteered'.
-----------------------------------------------
 {FD Additions NY4I}
TC_IMPROPERTRANSMITTERCOUNT = 'FD transmitters must be between 1 and 99.';
TC_IMPROPERARRLFIELDDAYCLASS = 'Field Day class must be A, B, C, D, E or F.';
TC_ARRLFIELDDAYIMPROPERDXEXCHANGE = 'DX Station exchange must be "DX".';

ny4i, 2016-02-24:
They are in my field day branch. Let me update it from master and you can merge.

ny4i, 2016-02-26:
These are now in all the language files for master.

closed 2022-02-05 by ny4i