TR4W/TR4W #108: Enable RITCLEAR under CWByCAT

n4af, 2016-02-17:
Allow RIT CLEAR to work like it does with WinKey (for supported radios).

ny4i, 2016-02-17:
I need some more detail. What does CWBYCAT have to do with RITCLEAR? 

Tom
Principal Solutions Architect, ITIL®v3 
Better Software Solutions, Inc.
727-437-BSS1 (727-437-2771)
@BSSI_Consulting

n4af, 2016-02-17:
User error, not set in the function keys

n4af, 2016-02-17:
Sorry Tom I did not have it set in the function keys.

73, Howie

http://n4af.net

closed 2016-02-17 by n4af

ny4i, 2016-02-17:
Clearly the program should have known what you meant to do and acted accordingly. :)

Glad it was a false alarm!

Tom
Principal Solutions Architect, ITIL®v3 
Better Software Solutions, Inc.
727-437-BSS1 (727-437-2771)
@BSSI_Consulting

n4af, 2016-02-17:
exactly so. stupid program :-)

73, Howie

http://n4af.net