TR4W/TR4W #110: TR4W loses sync with Wkey

n4af, 2016-02-18:
A report that Wkey loses connection to TR4W

n4af, 2016-02-18:
Not really sure of the root cause but force WK_RESET before initiating SENDCW
4.46.3

n4af, 2016-02-18:
4.46.3 breaks sent exchange.

n4af, 2016-02-18:
changed 4.46.3 to clearbuffer before initiating 
SendCrypticCWstring(Message); line 226 LogSubs1
This only flushes the cw buffer before an F key is processed.

ny4i, 2016-02-19:
I see you commented it out, but just a note that the RadioObject.SendCW is ony called by CWBYCAT. The idea is this comes under the guise of a radio command so by definition it would be CWBYCAT only.

n4af, 2016-02-19:
No.
Line 281 of LogCW calls him with Wkey:

if KeyersSwapped then                         // ny4i 4.
         begin
         InactiveRadioPtr.SendCW(Msg);
         end
      else
         begin
         ActiveRadioPtr.SendCW(Msg);  <<<<<<<<<<<<<<<<<<<<<<<<<
         end;

http://n4af.net

ny4i, 2016-02-19:
OK. Now I need to figure this one out. I thought I added SendCW when I added CWBYCAT. Let me look again.

ny4i, 2016-02-19:
Howie, I finished this and was testing with the WinKeyer. There were too many places where the FlushBuffer was called. The WinKeyer was not working. I could not tell if these were there before. I commented out three FlushBuffer calls that looked new. You can see my changes in the latest commit.

n4af, 2016-02-19:
Ok. Yes, that could lead to problems for sure.

http://n4af.net

n4af, 2016-02-19:
This will be 4.46.4

n4af, 2016-02-24:
i have asked N5AW for an update on whether this can be closed.

n4af, 2016-04-08:
closing- not able to reproduce this.

closed 2016-04-08 by n4af