TR4W/TR4W #111: Check if TR4W implements increasing CW speed via messages for CWBYCAT and WK

ny4i, 2016-02-20:
On the air, I heard a station send 5NN much faster than the rest of the exchange. I recall that TRLOG used to support that. I check the TRLOG manual and ControlF increases the speed by 6% and ControlS decreases it by 6 %. After the message is sent, the speed is restored to the original value (in case you had unequal speed up and speed down messages.

It looks like the serial CW implements this (I think( but I did not see support for CTRL-F and CTRL-S in SendCrypticCWString so I wonder if this works on the WinKeyer (and if implemented for that, on CWBYCAT as well). 

Does anyone know if you can increase the speed in a CW Message. Perhaps, you embed the CTRL-P, CTRLF in the command (I wonder if there is a better command we could use for up and down).

In any case, if this is implemented for serial keying then it should be implemented for CWBYCAT too (and the WK if not).

n4af, 2016-02-20:
Sure is for both.
Alt-P for the Fkey, hold down CTRL KEY and press P with ctrl still nown
press 'F' as in Fast. Do it 5 or 6 times. followed by '599' the the same
sequence of CTRL-P CTRL-S where same # of Slow as there were Fast.

73, Howie

http://n4af.net

ny4i, 2016-02-20:
Thanks. I tested that and it works. I will have to add something similar to the CWBYCAT case

ny4i, 2016-02-20:
The reason this was not obvious is that the CTRL-F and CTRL-S are not processed as part of the SendCrypticCWMessage. Instead, they are handled inline in the WK thread. The $06 is placed in the buffer and when the thread sees it, it changes the speed.

With this arrangement, I now know why the CW characters are sent as single bytes. In the K3 case, I send a single byte to the radio as the serial port handling is fast enough to process this fine. In the Icom case, it was a bit choppy. For a K3, I can easily just send the CWUp or down speed messages when I find them. For this to work on Icom, I would need to go back to sending a single character at a time. In my early testing, this did not work on my IC9100, but I will try again in case that was some other issue. Sending a character at a time makes it easier to stop in any event so that might be worth it. 

I will test CWUp and down with the K3 to see how it responds in the contest. I am using this in ARRL-DX with the WinKeyer and I like the up and down.

ny4i, 2016-02-24:
This is pending until Elecraft fixed what Wayne suggested was a bug.

ny4i, 2016-03-14:
I was able to make this work for the K3. This ended up in the Issue131 branch that is now in master.

ny4i, 2016-03-14:
Release Note: On the Elecraft K3, CWByCAT now processed the embedded speed up and speed down controls from a TR4W CW String. This uses the standard TR4W CTRL-P CTRL-F sequence as documented in the TR4W User's Guide.

closed 2016-03-14 by ny4i