TR4W/TR4W #112: When editing a QSO, old characters remain

ny4i, 2016-02-20:
I mistakenly entered 100 for the power in ARRL DX. I realized it right away and editing the logged QSO. I updated the power to KW and in the log window, the display shows KW0 (with I presume the 0 left over from before. 

I saw this behavior in another issue. It appears that the code to display the log must be not hitting a terminator. Check how this is displayed. Exporting the log to make sure it is set to KW and not KW0 should also occur.

ny4i, 2016-02-24:
uEditQSO.pas SaveQSOToEditableLog procedure is where the Save is processed when editing the log

ny4i, 2016-02-24:
What a bear to find...in SaveQSOToEditableLog in uEditQSO, when the fields of the EditableQSORXData record are updated, they are not initialized before hand. I am not sure if this is by design or not, but if I do something like this:
````delphi
{Power}
  ZeroMemory(@EditableQSORXData.Power, SizeOf(EditableQSORXData.Power));
  EditableQSORXData.Power := {Windows.GetDlgItemInt(eq_handle, FLD_POWER, lpTranslated, True);} GetDialogItemText(eq_handle, FLD_POWER);
````

it works. This code seems to have gone through multiple incarnations of the way it handled this. There are examples of 

````delphi
GetDlgItemText(eq_handle, FLD_QTHSTRING, @EditableQSORXData.QTHString[1], SizeOf(EditableQSORXData.QTHString) - 1));

// and

//  Windows.ZeroMemory(@EditableQSORXData.QTHString, SizeOf(EditableQSORXData.QTHString));

````

Both of those should initialize the field. It is possible that some fields were initialized but power was not.

Still testing...

n4af, 2016-02-24:
i also ran into this.if removing characters it was necessary to space over
them.

http://n4af.net

ny4i, 2016-02-24:
I have the fix for it. Testing out on other fields.

Tom

Principal Solutions Architect
Better Software Solutions, Inc.
727-437-2771

http://n4af.net

> What a bear to find...in SaveQSOToEditableLog in uEditQSO, when the fields
> of the EditableQSORXData record are updated, they are not initialized
> before hand. I am not sure if this is by design or [...]
lpTranslated, True);} GetDialogItemText(eq_handle, FLD_POWER);
>
> it works. This code seems to have gone through multiple incarnations of
> the way it handled this. There are examples of
>
> GetDlgItemText(eq_handle, FLD_QTHSTRING, @EditableQSORXData.QTHString[1],
SizeOf(EditableQSORXData.QTHString) - 1));
> // and
> // Windows.ZeroMemory(@EditableQSORXData.QTHString,
SizeOf(EditableQSORXData.QTHString));

ny4i, 2016-02-26:
Release Note: Previously, when a QSO was edited replacing an item with a shorter string (such as replacing a power of 500 with KW), the original, longer data was kept in the log record. In the aforementioned case, the new power would be KW0 (with the 0 left over from the 500). This has been corrected.
Release Note: Previously, if an incorrect exchange was entered, (such as WCF for ARRL Section), while the program provided a warning of the error, if the corrected item was a shorter length, the field would become the new data plus the left over characters of the old data (i.e., the new item is SF so the above case would leave SFF with the F from the original WCF).

closed 2016-02-26 by ny4i