TR4W/TR4W #113: user request for turn off split mode when changing bands.

ny4i, 2016-02-22:
I often RUN in split - I transmit on VFO B so I can use the remote tuning knob on VFO A as a big RIT knob on my Orion.
I S&P with split off (i.e., using VFO A) so I can use the same remote tuning knob on VFO A.

I frequently forget  to turn off split when starting to S&P.  Also, it would be nice if ALT-V and ALT-B would automatically
turn off split mode, because then I'm transmitting on the wrong band when I first find a station to call on the new band.

Ideally, the requested split-off feature would both sense the fast VFO tuning in CTRL-J config file and using the TAB key
to change mode from run to split mode.

Terry N4TZ

n4af, 2016-02-26:
Some rigs, like the K3 automatically turn off split on band change.

TR4W should already issue a warning to user when in Split.

BandUP or BANDDOWN will now force 
PutRadioOutOfSplit procedure to execute in LOGEDIT.
I have verified this should translate into a 'split off' command to the Orion
via LOGRADIO line # 1360

This will integrate into 4.46.5

n4af, 2016-02-26:
Also split Warning converted to language constants file using
TC_SPLIT_WARN so that it can go thru translation

N4TZ, 2016-02-26:
Seems to work on ALT-B / ALT-V band changes

Ideally, the requested split-off feature would both sense the fast VFO tuning in CTRL-J config file and using the TAB key  to change mode from run to split mode.

Terry N4TZ

n4af, 2016-02-26:
Ok, I will leave this open for the ctrl-j enhancement, but this is not our
top priority.

For enhancements, mine would be SO2R box support.

http://n4af.net

N4TZ, 2016-02-26:
TAB (SHFT-TAB) key to turn off split would probably just require the same code line you added for CTL-B / CTL-V
Terry N4TZ 

 

http://n4af.net

n4af, 2016-02-26:
Yes, that would be easy enough to do.

73, Howie

http://n4af.net

n4af, 2016-02-26:
Added force split off via CTRL+- 
control and minus key
4.46.8

closed 2016-02-26 by n4af

N4TZ, 2016-02-26:
Not sure what you mean about split off with CTRL and minus key?The version I downloaded earlier today works with ALT-V / ALT-Bbut does nothing with CTRL and minus.
BTW, while trying the CTRL and minus to see what happens,I also tried just the minus key (to enter split) - and nothing happened.  In the old version I have used, when the focus is inthe callsign area, typing a minus sign opens a message for enteringsplit transmit frequency.  This is not working in the version I'm trying. Pretty important with the upcoming fone contest!
Terry N4TZ

n4af, 2016-02-26:
http://tr4w.net/4.46/tr4w_setup_4_46.8.exe

http://n4af.net

N4TZ, 2016-02-26:
I download 4.46.8 - CTRL minus turns off split (still would like the TAB key (mode change) to do that
BUT, trying to use the minus key to enter split tx frequency doesnot work in 4.46.8, either
Terry N4TZ 

 

 http://tr4w.net/4.46/tr4w_setup_4_46.8.exe

http://n4af.net