TR4W/TR4W #114: LASTCQFREQ fails to work

n4af, 2016-02-24:
This was a serious problem in ARRLDX. 
I would go to a spot and want to return to run frequency only to have the command do nil.

Turning on debug, LASTCQFREQ works. Compiled TR4W works. Make TR4W build using FULL.NSI (which reads in ancillary files and compresses the exe) produces tr4w that no longer QSYs on the command.

n4af, 2016-02-24:
Ok. Another 'I didn't know'.

*IF* BANDMAP DISPLAY CQ is not turned on this does not work.

Q: is that acceptable ? Gavin ?

n4af, 2016-02-24:
Line 332 of LogSubs1 is doing this.
Do we need this line ?
if BandMapEnable and BandMapDisplayCQ then

n4af, 2016-02-24:
If Gavin agrees, I propose scrubbing the above test and rolling the change into 4.46.5

BTW, not thinking this is an example but I have been burned by the different code paths on **Fx CQ vs Enter Key CQ**
Might be worth remembering.

73, Howie

n4af, 2016-02-24:
4.46.3 built - ready for support testing

gm0gav, 2016-02-25:
Hi Howie,

never used that command, seems a bit odd but that command is obviously 
relying to the spot being there to take you back.

That command probably needs a rewrite to create its own variable of  
current cq freq when you cq, then use that as the freq to return to

73 Gav

n4af, 2016-02-25:
Note -

This went out as 4.46.3 for testing. See comment above your last post.

https://groups.google.com/forum/#!topic/tr4w-support/Z-WIVGxvgks

closed 2016-03-01 by n4af