TR4W/TR4W #118: Exception code: 0xc0000005 in TR4W if internet connection to dxc drops

n4af, 2016-02-28:
Access violation may occur when internet connection is lost:

Faulting application name: tr4w.exe, version: 0.0.0.0, time stamp: 0x2a425e19
Faulting module name: tr4w.exe, version: 0.0.0.0, time stamp: 0x2a425e19
Exception code: 0xc0000005
Fault offset: 0x00001e95
Faulting process id: 0x11c4c
Faulting application start time: 0x01d171cdb3a4f194
Faulting application path: C:\Program Files (x86)\TR4W\tr4w.exe
Faulting module path: C:\Program Files (x86)\TR4W\tr4w.exe
Report Id: bbe37fd6-1a6c-4745-bba8-071b212fe2b7
Faulting package full name: 
Faulting package-relative application ID:

ny4i, 2016-02-28:
One way to test this would be to use the CC User program that connects to a cluster on your behalf. You then connect to a port that CC User offers. You can then use pskill (in SysInternals) to kill the CCUser task. This should provide a simulation of a dropped connection. 

You cold always unplug your Ethernet cable :).

I have not tried this but I will.

ny4i, 2016-03-01:
Well that test did not work correctly.

Here is what I did,

I obtained the VE7CC CCUSER program from http://www.bcdxc.org/ve7cc/ve7cc_win7.zip

I then obtained the SysInternals Tootls suite to gain access to the pskill program. pskill will immediately stop a Windows program. 

SysInternals is available here: https://download.sysinternals.com/files/SysinternalsSuite.zip

In CCUser, under the Configuration ...Port/Logging menu, I enabled the Logging Program Connection checkbox and used the default port of 7300.

In TR4W, I had the program connect to localhost:7300.

After I verified I have a connection, I used pskill ve7cc and the program was immediately shutdown. I received a message from TR4W stating, "An existing connection was forcibly closed by the remote host"

I suspect that even though I am killing the CCUser program, the socket is still closing with some notification to Windows.

ny4i, 2016-03-01:
My next test was to connect to the cluster directly from TR4W, and after ensuring WiFi was turned off on my laptop, I unplugged the Ethernet cable. This did not cause any fault either. After a short period of time, I did receive the same message from TR4W,  "An existing connection was forcibly closed by the remote host"

ny4i, 2016-03-01:
I did note that we do not try to reconnect. I would think we would want to try to reconnect periodically.

ny4i, 2016-03-01:
So at this point, I have to say I cannot reproduce the problem. If anyone else has some ideas on how we can recreate, please post here. -- NY4I

n4af, 2016-03-01:
Thanks for your efforts Tom.
closing

closed 2016-03-01 by n4af