TR4W/TR4W #124: Tailend Key not working

gm0gav, 2016-02-28:
Confirmed that Tailend key is not working in 4.46, TRLOG manual states it works like this:

To respond to a tail-ender, you can use the TAIL END KEY. When you
are finished with your current QSO, hit the TAIL END KEY, which causes
the TAIL END MESSAGE (normally an R) to be sent, after which you can
start to enter the call of the tailender. As you enter the call, it will be
transmitted. Pressing <Enter> will complete sending the call and also send
the exchange. The previous QSO will now be logged and the new one set up
correctly. The default TAIL END KEY is ].
When entering a callsign, you can use ? as a character. Any instances
of ? must be changed to a legal callsign character before TR-LOG will let you
log the contact

gm0gav, 2016-03-08:
Tailend message and key commented out by     //n4af 4.41.5
Best if Howie looks at this, I am unable to see a reason why it should be commented out.

n4af, 2016-03-08:
It has been  well over 20 YEARS since I have  heard the tail end key used.
It is hard enough to get a station to listen after signing without CQ. This makes zero sense to me.
(Even in Eu Sprint)

WHAT CONTEST IS IT USED IN ???
WHAT TR4W level supports it ???
I cannot see it works in 4.248 which is the level Dmitry gave us the source.

Right now this does not register on my to-do list.

73, Howie N4AF

ny4i, 2016-03-30:
I saw some recent activity on this from our Russian friends. if they have a real need, I don;t see way we do not put it back. Maybe we can ask the tr4w list if anyone uses TailEnding anymore? I guess we would have heard about it more if it was missed but if it fills a need, I can see adding it back.