TR4W/TR4W #125: - key in call window does nothing

ny4i, 2016-02-29:
From Terry N4TZ,

> 4.46.9 does not respond to split frequency request (the minus key)
> to enter split frequency.

ny4i, 2016-02-29:
From the TRLog manual,

alt-N

> The alt-N command will put your Kenwood or Yaesu radio into split
> mode and set the transmit VFO to the frequency entered. You can
> enter just the last three digits of the frequency in most cases. You can
> also activate this function with the single dash key, allowing you to
> enter the frequency by using only the numeric keypad (in NUM LOCK
> mode).

Did the - part get dropped somewhere along the way?

n4af, 2016-03-02:
Starting 4.46.13 CTRL plus the minus key will TOGGLE split.
Generally available 4.47

ny4i, 2016-03-03:
I'm curious why it can't be -?

n4af, 2016-03-03:
maybe http://www.delphigroups.info/3/2c/8807.html

When you have the answer, I would like to know which of the string theory 11 dimensions we occupy ?

n4af, 2016-03-08:
Changed window menu item to again support the dash (minus) key
fixed 4.47.3
NOTE: menu selection for TX FREQ is now under COMMANDS

ny4i, 2023-04-06:
Resurrecting an old issue. I see now why just using the - key is not a reasonable way to do this since the program uses it as an accelerator. I suspect if instead of an accelerator, it was handles in the keydown of -, we could make it context sensitive. I would think that a - pressed only when there is nothing in the call or exchange window would be the way to go to the set TX split frequency. 

I ran into this as I was adding a new "contest" type of POTA. POTA parks have a dash in the name of the park like K-1234. We can just tell the user to not press the - but it would be better to show up as we type it. 

I will look further into making the dash something besides an accelerator in the RES file.

ny4i, 2023-04-08:
I removed the - key as an accelerator and handled it in the callWindowKeyDown procedure. This way we can use - in other places where it makes sense.