TR4W/TR4W #129: CQ Inactive sends on two radios

n4af, 2016-03-02:
Active radio continues to send when CQ Inactive is initiated.
This places the SO2R station in multi category.

n4af, 2016-03-02:
ALT-D is Ok.

n4af, 2016-03-02:
Wkey works ok.

ny4i, 2016-03-03:
I think I have this working. Can you share the scenarios that caused it to do this? 

I was sending CQ on active and now when I send CW on inactive,the active radio stops right away.

ny4i, 2016-03-03:
Howie, The Pull request I created seems to get closer to this. Please let me know if this works properly.

n4af, 2016-03-03:
Hi Tom -

To the best of my knowledge this was fixed some time ago.

CQ on active rig and then press CQ Inactive key. The main cq should stop
immediately.

73, Howie

http://n4af.net

n4af, 2016-03-03:
ok, my mistake. I did recently open this, but fixed it pretty quickly in the latest 4.46.13

closed 2016-03-03 by n4af

ny4i, 2016-03-03:
I worked it because I could press CQ inactive and CW active and they both
were sending.

It works now.

Principal Solutions Architect
Better Software Solutions, Inc.
727-437-2771

http://n4af.net

n4af, 2016-03-03:
ok. seems fine. it is up as 4.46.14

You were up late ?

73, Howie

http://n4af.net

n4af, 2016-03-03:
Best I can tell, by exploiting CWByCAT_Sending,
AUTOSEND with CallTerminate is working...

http://n4af.net