TR4W/TR4W #130: Escape fails to halt CQ Inactive

n4af, 2016-03-02:
Only fails using CWByCAT

ny4i, 2016-03-02:
OK. So here is my two cents. CWBYCAT is a harder implementation for the various items involved. This was working but then we had some issues with Two Radio Mode. Because of that, we started adding more spaghetti to handle exceptions to the main data path processing. So, here is what I propose…

If after initialization you are in two radio mode and CWBYCAT is active, I am going to display a message that CWBYCAT and Two Radio Mode have not been fully implemented and may not work in all cases. We recommend this not be used in two radio mode.

This gives me some time to rethink how I am implementing this and address all the cases. I want to make sure single radio mode is bullet proof but keep in mind two radio mode.

I also had a thought that the program uses a single WinKeyer versus two WinKeyers. I know it flips the outputs. While that is OK if I am running a standalone WinKeyer box, on things like my microHam mkII (which has an inbuilt winkeyer), the program switching the WK would not work. This is another case of CWBY and two radio mode incompatibilities. If I have the same radios types, this should be fine, but if I am trying to use CWBY on one radio and the WK on the other, all kinds of things break. I’m not sure I can fix that unless we establish support for two unique WinKeyers (or have a WK mode with single/dual.

Bottom line…I propose to turn off CWBYCAT of two radio mode is on (or we can jkust display the message and let the user assume the risk).

Tom
Principal Solutions Architect, ITIL®v3 
Better Software Solutions, Inc.
727-437-BSS1 (727-437-2771)
@BSSI_Consulting

n4af, 2016-03-02:
agreed. i see a better way to do the conventional escape tree but there are
other problems

73, Howie

http://n4af.net

ny4i, 2016-03-02:
OK. Well let’s get everything checked in and I will work on this tonight. I will most likely go back to the esc code in an earlier version when I first did CWBC and just add what we have learned since.

My test case will be CWBC, WK and serial CW.

I have been giving the ESC some thought too. MY approach is to check if we are still sending even in CWBC.

Thanks,

Tom
Principal Solutions Architect, ITIL®v3 
Better Software Solutions, Inc.
727-437-BSS1 (727-437-2771)
@BSSI_Consulting

n4af, 2016-03-02:
ok. i have a fix for the cq inactive that attempts to do the escapes.
will put it up asap

73, h

http://n4af.net

n4af, 2016-03-02:
latest 4.46.13 appears to remedy
This is now git master

n4af, 2016-03-03:
CWBC_Issues is now 4.46.14 and current master

ny4i, 2016-03-03:
Current test: In S&P mode, send CQ on Inactive radio, hit ESC to stop CW on Inactive and program returns to CQ Mode. But if sending code on active radio (still in S&P mode), ESC just stops the CW.

It appears that if the program is checking if CW is being sent, it may not be checking the inactiveRadio.

Perhaps check if the InactiveRadioPtr.CWBTCAT_Sending is true or ActiveRadioPtr.CWBYCAT_Sending is true and simply stop sending.

If neither one of those are true, and we are in S&P mode AND the Exchange window, simply return to the call window. If we ESC in the call window and it is blank and we are not sending, then return to CQ Mode. 

This seems to be easier logic than written.

We need a flow of how we want this to work and just rewrite the ESC procedure. 

Now that I added the CWBYCAT_Sending procedure, this should be easier. We can use WKBusy too.

n4af, 2016-03-03:
Build of issue130 correctly handles switch from inactive back to active , calling in s&p and hitting escape.
It does not correctly handle a station answering your cq , entering his call, pressing enter, hitting escape (instead of just stopping sending it clears the window)

ny4i, 2016-03-03:
OK. I will tackle that tonight.

I wonder if you might be able to give me a list of all the things I should test. As I am new to SO2R, I am sure I will miss steps.

I also need to understand Auto-Terminate a little better. Maybe we need a TeamViewer session :)

Thanks,

Tom
Principal Solutions Architect, ITIL®v3 
Better Software Solutions, Inc.
727-437-BSS1 (727-437-2771)
@BSSI_Consulting

n4af, 2016-03-03:
well, keep in mind some of this is important and some 'not so much'
Typically Run/S&P Active one first press is top priority

trying to organize columns by importance, a table might be organized

columns:                                            *RUN active S&P active
RUN inactive S&P inactive*
rows         *WK/one press*
*                WK/two*
*                WK/3*
*               CWBC1*
*               CWBC2*
*               CWBC3*
*               Serial 1*
*                         2*
*                         3*
*               Parallel 1*
*                          2*
*                          3*

http://n4af.net

ny4i, 2016-03-04:
Something like this... https://github.com/n4af/TR4W/wiki/Testing-Matrix-for-CW-Sending

ny4i, 2016-03-10:
Well the ESC processing is not working again.

I am working in this area to make this code more bulletproof. 

My issue now is I am sending CW in SPMode and the CW stops but it returns to CQMode. The Esc counter stuff is off so I propose I just abandon that and do the processing the way the code did before CWBC.

I am working in this area so if anyone needs to modify the ESC code in MainUnit, please let me know.

ny4i, 2016-03-11:
After giving this some thought today, I gutted the code added to the Escape_proc for CWBC. My error was adding code to handle the specific case. What I now did was add code to places that were called normally like PTTOff. In serveral places I tried to add CWBC code and not disturb any other code. For example, there was code that read like this in Escape_proc:
````delphi
 if ((ActiveMode = CW) and
     ((CWThreadID <> 0) or (wkBUSY = True) )) or
    ((ActiveMode in [Phone, FM]) and (DVPOn = True)) then
  begin
...
````

The code handled in this block is what made the ESC processing work to stop the CW. I updated this to accept that CWBC is as valid a CW sending mode as any other.
````delphi
 if ((ActiveMode = CW) and
     ((CWThreadID <> 0) or wkBUSY or pRadio.CWByCAT_Sending)) or
    ((ActiveMode in [Phone, FM]) and (DVPOn = True)) then
  begin
````

Now we will stop the CW properly in later procedures by checking CWBC_SENDING and others. So, the culprit here was trying to bolt on CWBC without fully-integrating it into the CW sending procedures that already existed.

ny4i, 2016-03-20:
My testing shows this works OK.

I can use ESC and stop CW if I am calling CQ Other, or using Alt-D after a spacebar.

Please update this case if there are any other issues with stopping CW.

It would be good to check the WInKeyer to make sure I did not break anything.

If all works, then close this case please.

closed 2016-03-20 by n4af