TR4W/TR4W #131: CWBC autosend broken

n4af, 2016-03-02:
Drops last character past autosend character count

n4af, 2016-03-02:
i am extending this to Wk
Broken in 4.46.12

n4af, 2016-03-02:
Pushed an amended copy of 4.46.13 up to master that fixed Wkey.
New copy 4.46.13 also uploaded to 4.46 directory to replace the original that I broke.

This does NOT fix CWByCAT which we do not recommend for SO2 at this time.

ny4i, 2016-03-08:
I'm close to deciding if we can reliably do AUTOSEND with CWBC. I spent some time looking at how this works with the WK and SerialCW. Technically, AUTOSEND works fine, but AUTOTERMINATE is the one that presents the issue. I shaved 500 mS off the time to get the TX status but it may still not be enough to work properly. 

I am still working on it and should have a decision soon.

If it turns out as such, we may need to state that AUTOTERMINATE is not compatible with CWBC and state as such (via ShowMessage(TC_CWBCWITHAUTOTERM) if both settings are enabled.

More to come...

ny4i, 2016-03-14:
Release Note: This now works properly in CWByCAT mode. Note that while you can also use AutoTerminate, there is a noticeable delay over using the WinKeyer or serial CW. The issue is I have to poll the radio. Unless the radio polling is put to its own thread, the lag is noticeable. If you are a user of AutoSend, there is no delay. But if your operating requires AUTOTERMINATE, I recommend you do NOT use CWByCAT.

closed 2016-03-14 by ny4i