TR4W/TR4W #135: Add new dialog box requesting split TX frequency

n4af, 2016-03-04:
TR4W should put up a TX FREQ dialog box.
Then put the radio into split and set the transmitter to the entered frequency for the operator.

This was apparently dropped somewhere between 246 and 248 and is not in the source code we have.

n4af, 2016-03-04:
"In Classic TRLog, and Dmitri's implementation of TR4W,
when an operator is S&Ping, finds a station transmitting on 7060
saying "listening on 7234," the operator then hits the minus key
on the computer keyboard which brings up a little box in which
you enter 7234 and hit enter, which puts you in split mode,
with your receiver still tuned to 7060 but now your transmitter
is on 7234 ready to call the DX station."

n4af, 2016-03-05:
picked up in 4.47.1, which is up for testing

ny4i, 2016-03-10:
This works but there is something strange.

1. Put your cursor in the call window
2. Enter -
3. The TX dialog box pops up as expected.
4. Enter the TX frequency (I entered 200 as I was on 20m)
5. VFO B changes to 14200
6. Now hit - (while in call window)
7. Nothing happens.
8. Hot - again, and now the TX frequency dialog appears.

It seems like that first - after the frequency is changed does not happen.

n4af, 2016-03-10:
I cannot seem to reproduce this.
If this is on the K3, the only indicator split has been reset is the split light-- vfo b still indicates the split freq. but WINDOW pulldown RADIOx should show the radio going out of split ?

ny4i, 2016-03-10:
Split is not reset. The error is just that the first - hit after steps 1 - 5 is not processed. I have to hit - twice for the dialog to come up. Go through this step by step if you can. -- Thanks

n4af, 2016-03-17:
I think it is working as designed.

If IN SPLIT it turns it off.
If OUT of Split it goes into the dialog

closed 2016-03-17 by n4af

ny4i, 2016-03-17:
I agree. The menu may need to indicate this is a toggle by updating from Split On to Split Off though to be consistent. The latest I saw says Split On/Off. I wonder if it would be easy to set the text to the correct state. Also, in 4.47.1, the accelerator key said Ctrl- but I think I saw that updated to just dash somewhere.

Thanks,

Tom
Principal Solutions Architect, ITIL®v3 
Better Software Solutions, Inc.
727-437-BSS1 (727-437-2771)
@BSSI_Consulting

n4af, 2016-03-17:
Hi Tom -

Ok, I changed the verbage on the menu to 'Toggle Split'.

It will piggy back the next commit.

73, Howie

http://n4af.net