TR4W/TR4W #136: Split dialog needs to toggle split status

n4af, 2016-03-05:
Control and Minus key should invoke the Split dialog initially but if invoked again take down Split without having to enter a zero.

n4af, 2016-03-05:
Fixed 4.47.2

n4af, 2016-03-07:
Clarification on how this works:
control and minus keys initially set split; to enable split for the inactive radio the frequency must be preceded with a minus sign.
To explicitly go out of split , enter '0' for active radio or '-0' for inactive radio.
To force either out of split, subsequent control minus should suffice.

ny4i, 2016-03-08:
Just to clarify, do we still not know where the - is getting consumed by the application if the user just enters -7003 in the call window? At this point, I'm just curious what is eating the dash. I do like the way TRLOG handled this where there was no extra dialog box and one just entered -7002 but I also like the idea you added to being able to set this for the Inactive radio too.

n4af, 2016-03-08:
Hi Tom -

Our interface to the user begins at WM_CHAR in main callback.
None of the WindowsMangement routines detect the dash key.

Something like the '=' key works.

73, Howie

http://n4af.net

ny4i, 2016-03-08:
What's odd is I knew this worked. To prove this, I downloaded http://tr4w.net/4.32/tr4w_setup_4_32.0.exe (4.32). I run this and while sitting n the call window, I hit a single dash ( - ). As I suspected, the transmit frequency dialog comes up. 

I remember this being removed as I mentioned previously when you changed this so the dash was not acted upon while in other windows. I can diff the code to find out what happened.

ny4i, 2016-03-08:
This works up until 4.43 but not in 4.44. Diffing between those two...

ny4i, 2016-03-08:
Wow...this is a puzzler. I cannot figure out what in the diff caused us to lose the - key. I did find the commit that did this.

https://github.com/n4af/TR4W/commit/833e8aa24b6e351d30fd4b00daacf383e0da18e4

Maybe it will jog your memory...

n4af, 2016-03-08:
> 833e8aa

what rel of windoze u running ?

http://n4af.net

ny4i, 2016-03-08:
Windows 10. I can do the same tests on XP or Win 7 to check. The thing is that on version 4.43 and the same OS, the dash key works fine.

n4af, 2016-03-08:
put a stop on wm_char in 4.43

http://n4af.net

n4af, 2016-03-08:
if (Msg.HWND = wh[mweCall]) then CallWindowKeyDownProc(Msg.wParam);

http://n4af.net

ny4i, 2016-03-13:
Howie - Can you check something for me. 

I used the code in master with AUTOSEND set to 3 characters. AUTOTERMINATE is also true but that can be false and it still behaves as I describe below...

When I hit the third character, the program is switching to the Other radio (making it active) and sending CW Memory F1. At first I thought it was my branch, but the code in master is doing it too.

n4af, 2016-03-13:
Hi Tom -

I QSL.

As I recall, line 723 of the TR4W.DPR is doing that on
'backtoinactiveradio...'

http://n4af.net

n4af, 2016-03-13:
i think this is hooked to the wrong issue...

ny4i, 2016-03-13:
Howie,

That line is in this block of code:
````delphi
if (Msg.HWND = wh[mweCall]) or (Msg.HWND = wh[mweExchange]) then
          begin
            if Msg.wParam in [VK_F1..vk_f12] then ProcessFuntionKeys(Msg.wParam);
            if Msg.wParam = VK_F4 then goto NoTransMess;
            if Msg.wParam > 40 then goto TransMess;
            if Msg.wParam = VK_RIGHT {39} then if Msg.HWND = wh[mweExchange] then TryPutSpaceinExchangeWindow;
            if Msg.wParam = VK_PRIOR {33} then ProcessMenu(menu_cwspeedup);
            if Msg.wParam = VK_NEXT {34} then ProcessMenu(menu_cwspeeddown);
            if Msg.wParam = VK_SPACE {32} then if Msg.HWND = wh[mweCall] then
              begin
                SpaceBarProc2;
                if ActiveRadioPtr^.CWByCAT then BackToInactiveRadioAfterQSO;
                goto NoTransMess;
              end;
````

It looks like the line:  if ActiveRadioPtr^.CWByCAT then BackToInactiveRadioAfterQSO; is only executed if the spacebar is hit and the current window is the call window. That is not the case here as I have typed something in the call window.

BTW, this check should is IsCWBYCATActive as that tests the radio model too).

n4af, 2016-03-13:
ok, we are talking about two different things.
i am talking about ALT-D not working.

73, Howie

http://n4af.net

ny4i, 2016-03-13:
I thought the error I found might be related which is why I put it here.Alt-D is a different issue.

This is what I wrote earlier...

> I used the code in master with AUTOSEND set to 3 characters. AUTOTERMINATE is also true but that can be false and it still behaves as I describe below...
>
> When I hit the third character, the program [...]

Can you see if it works the same for you? If so, then the current code is master is unusable for 2 radios. The InactiveRadioCQ is really what this looks to be doing which is why I thought it might be related.

ny4i, 2016-03-13:
I realize I may have added this to wrong issue. I was looking at the commits and I saw this one but the issue could have been there back from the Alt-D change you made. -- Thanks

n4af, 2016-03-13:
Ok, I guess I had alt-d on the brain.

Will check...

http://n4af.net

n4af, 2016-03-13:
i am gonna assume we are talking about cwbc ?
Does it do that under 4.47.5 ? I think that is the master level ?

Yes, it does fail.
I left some residual code in working on cwbc alt-d that is causing the
problem

73, Howie

http://n4af.net

n4af, 2016-03-13:
Hi Tom -

Might be easier if you are actively working on something to make the change
yourself--
afaik, the problem is all in MainUnit

*procedure CallWindowKeyDownProc(wParam: integer);*

down into the routine the block within ( i have already commented out the
bad blocks)

  if Key <> StartSendingNowKey then
      begin
        if IsCWByCATActive then
           begin // Send the character now - No buffering
           if (length(CallWindowString) = AutosendCharacterCount)  then
//n4af 4.46.12
              begin
              ActiveRadioPtr.SendCW(Key);  // start sending if = autosend cc
        {      ActiveRadioPtr.SendCW(Chr(242));    // comment out down to
              end;
          if (length(CallWindowString) > AutosendCharacterCount) then  //
hit additional key(s)
           begin
          ActiveRadioPtr.SendCW(Key);
          ActiveRadioPtr.SendCW(Chr(242));
           wait:
          if i > 5 then exit;    // half second limit on loop
           if autocallterminate and not activeradioptr.CWByCAT_Sending then
           processreturn       // autosend switch from call to exch window
when sending donecw
           else
           begin
           inc(i);
           sleep(100);   // wait on sending completion
           goto wait;  }    // here but see below
          //  end;
           end;
          end
        else if wkActive then
        wkSendByte(Ord(UpCase(Key)))
        else
          begin

          //localMsg := Format('After AutoChar - key = %s;', [key]);
          //AddStringToTelnetConsole(PChar(localMsg),tstAlert);

          CPUKeyer.AddCharacterToCWBuffer(Key);  //
          end;
      end;
      EditingCallsignSent := False;
    end;
  end;
  //  CallsignsList.CreatePartialsList(CallWindowString);
  p := wh[mwePossibleCall];
  c := wh[mweCall];
  if not InsertMode then EditSetSelLength(c, 1);

  itempos := SendMessage(p, LB_GETCURSEL, 0, 0);

  if Key = PossibleCallLeftKey then dec(itempos);
  if Key = PossibleCallRightKey then inc(itempos);
  if itempos = -1 then itempos := 0;
  SendMessage(p, LB_SETCURSEL, itempos, 0);

  itempos := SendMessage(p, LB_GETCURSEL, 0, 0);

  if Key = PossibleCallAcceptKey then

    if SendMessage(p, LB_GETCOUNT, 0, 0) > 0 then
       begin
       PutCallToCallWindow(LogSCP.PossibleCallList.List[itempos].Call);
       end;
  {  if activeradioptr.CWByCAT_Sending then     // this also needs to come
out
       begin
       Sleep(1500);
       BackToInactiveRadioAfterQSO;
       end;   }

http://n4af.net

ny4i, 2016-03-14:
When you put the radio in Split mode with the - key, should the QuickDisplay message of You are in Split Mode appear? It does not now...

n4af, 2016-03-14:
Hi Tom -

The message is event driven. It will appear if you start to, say, enter a
call in the call window.

73, Howie

http://n4af.net

closed 2016-03-17 by n4af