TR4W/TR4W #141: Bandmap does not clear all spots cleanly

gm0gav, 2016-03-08:
Clearing all spots leaves spurious characters for a short period

gm0gav, 2016-03-08:
Fixed in  4.47.4, Killfocus procedure added to cleanly move focus away from bandmap

ny4i, 2016-03-10:
Gav - There is some artifact left when dealeting a single spot though now.

ny4i, 2016-03-14:
This is what I saw after I deleted a single spot.

[image]

I'm guessing but it seems that the code to delete a spot and atomically update the number of items in the spots array and then refresh only the items int he spots array is not exactly synchronized. This is an very event driven process so I would think there has to be some concurrency controls added (mutex, TCriticalSection, etc).

gm0gav, 2016-03-14:
Noted the same, will resolve this week

ny4i, 2016-03-30:
I'm taking a shot at this...

ny4i, 2016-03-30:
I'm punting this for now. If I work in this too long, I will rewrite the whole thing as a real Delphi form with a real TList object filling the window. :)

commit b739bc36 (2016-04-01): Fixes #141 and #105

#141 Added Kllfocus procedure to cleanly move focus from Bandmap,
resolves issue of rubbish left in bandmap after clear all spots.

#105 Resolved issue of bandmap flipping band on every ptt event. Now
only changes the bandmap band when the frequency changes.
closed 2016-04-01 by n4af