TR4W/TR4W #144: Yaesu FTDX5K randomly activates split when clicking spot from bandmap

gm0gav, 2016-03-12:
4.47.4 appears same for FTDX5K and FTDX9K settings, 50% of time and seemingly randomly puts rig into split

n4af, 2016-03-12:
Pop-a-gopher.

Split is now triggered by entering a minus sign.

Is there a minus sign in the spots that is doing this ??

73, Howie

http://n4af.net

gm0gav, 2016-03-12:
Howie,

that may well be it, caused a two radios on same band incident at my end 
which is never good!

73 Gav

ny4i, 2016-03-12:
I would a minus sign in the spot do that?

Principal Solutions Architect
Better Software Solutions, Inc.
727-437-2771

n4af, 2016-03-14:
I will need more information on what is causing this.

gm0gav, 2016-03-30:
From what I can see issue may be in 
procedure RadioObject.SetRadioFreq(Freq: longint; Mode: ModeType; VFO: char);

There is no reference to FTDX5K in this procedure, only older Yaesu radios.

ny4i, 2016-03-30:
I will check. There are multiple paths of the code depending if it was a newer radio.

ny4i, 2026-11-04:
Already fixed in the cross-platform TR4W (TR4W/TR4W-D12, FreePascal/Lazarus): the spot click puts the radio out of split unless the spot carries a split frequency (uBandmap TuneRadioToSpot); the FTDX5000 check is in the bench queue. Closing here; any new report goes to TR4W-D12.

closed 2026-11-04 by ny4i