TR4W/TR4W #145: With CWByCAT, on an Icom radio, sometimes onlky first letter is sent

ny4i, 2016-03-14:
This is a new issue from the Issue 131 changes fixed in 4.47.6. You can send one string but sometimes, the next function key will only send the first letter.

ny4i, 2016-03-14:
As I think I am the only one that uses an Icom radio as radio 2, you will have to take my word for it that this works. But please do make sure I did not break Kenwood or K3 CWBC.

closed 2016-03-14 by ny4i

n4af, 2016-03-14:
Hi Tom -
I have checked to ensure so2r WK remains ok. It does.
I will open some issues on so2r cwbc but not sure how practical it is to address them. 
On the one hand it does not seem like the radio firmware will ever provide the timing functionality of the  WK--OTH, when I key my amp via cwbc, timing seems ok on ptt engage/disengage ??

PR #148 (2016-03-17): Fixed issue where Icom radios was dropping a message.
The closes #145.