TR4W/TR4W #147: Add dynamic debugging to a file that can be enabled by users

ny4i, 2016-03-14:
Add a debugging facility to debug to a file such that the user can enable it at runtime. The logging should be level selectable (like Log4J, FATAL, ERROR, WARN, DEBUG, TRACE). The log should be to a file. Make use of existing Delphi logging frameworks if possible.

ny4i, 2016-03-14:
In Issue #145, I started to add some code for Debugging Icom messages (hex to displayable hex strings). This is not the full features implementation this calls for however. You still have to enable the NEWER_DEBUG Boolean to turn this on and it logs to the telnet window.

ny4i, 2020-07-31:
There is now a real-time configurable debug facility. This is controlled by a CTRL-J option called DEBUG LOG LEVEL. Set it to NONE, TRACE, ERROR, WARN, INFO, DEBUG or TRACE. The file is written to the log directory and programmers call it by stating logger.DEBUG(string) or logger.Fatal(s) or logger.trace(s).

n4af, 2020-07-31:
I would like to update DESCRIPTION in ctrl-j for DEBUG LOG LEVEL.

Could you provide some addl verbage ?
I see debug log level trace builds TR4W.LOG
info on entry:
7/31/2020 *180777 [11228]* debug TR4WDebugLog  - CrypticCWString = [CQ CQ
\  \ TEST]
What would the highlighted # reference ?
Since default DEBUG also writes to log, what is diff between TRACE and
DEBUG ?
Could you elaborate on 'programmers call it by stating
logger.DEBUG(string)' ?

Thanks, Howie

ny4i, 2020-07-31:
Sure.

The text would be something like this:

TR4W has a runtime logging facility. In diagnosing problems, it is sometimes helpful to have a log of the activities. If you experience an issue, the TR4W support team may ask you to enable a higher logging level and send the log file created (TR4W.LOG). There is no reason to enable this unless requested.

 Here is a breakdown of the fields:

> 7/31/2020 *180777 [11228]* debug TR4WDebugLog - CrypticCWString = [CQ CQ
> \ \ TEST]

Date  = 7/31/2020
Time = 180777
Thread ID = [11228] 
Log level = debug
internal logger instance = TR4WDebugLog
Log text = CrypticCWString = [CQ CQ \ \ TEST]

For the programmers (you, me, anyone else):

The above message can be logged in m multiple ways:

logger.debug(‘CrypticCWString = [‘ + s + ‘]’);

But I also call logger.debug(s) in the procedure DEBUGMSG;

Search on DEBUGMSG and you will see a call to logger.debug.

Search on logger. And you will find other examples. An example of the different levels is as follows:
\
If the log level is set to FATAL, then calls to logger.debug will not be logged. Only calls to FATAL will be logged.

If I set the log level to ERROR, the any calls to logger.error or higher (logger.trace) are logged. It is a hierarchy of log levels so the any logger.<level> call is actually logged if at the log level or higher.

This uses Log4D which is based on the concepts of Log4J.

Make sense?

Tom

> Could you elaborate on 'programmers call it by stating
> logger.DEBUG(string)' ?

Principal Solutions Architect
Better Software Solutions, Inc.
727-437-2771

closed 2021-05-02 by ny4i