TR4W/TR4W #150: CWBC - Escape does not kill CQ Inactive

n4af, 2016-03-14:
Cannot kill a CQ on the inactive radio

ny4i, 2016-03-16:
I have some time tonight to get to this and the other CWBC bug

ny4i, 2016-03-17:
The problem here appears to be an issue of which radio we stop sending on. In the code in LogCW, this occurs:
````delphi
procedure FlushCWBuffer;

begin
//  CPUKeyer.PTTUnForce;
  if IsCWByCATActive then
     begin
     DebugMsg('Flushing CWBuffer - Stop Sending on ActiveRadio CWBC');
     ActiveRadioPtr.CWByCATBuffer := '';
     ActiveRadioPtr.StopSendingCW;
     end;

  tAutoSendMode := False;
  CPUKeyer.FlushCWBuffer;
//  if wkActive then wkClearBuffer;    // Gav    remove
end;
````

The code:
````delphi
 if IsCWByCATActive then
     begin
     DebugMsg('Flushing CWBuffer - Stop Sending on ActiveRadio CWBC');
     ActiveRadioPtr.CWByCATBuffer := '';
     ActiveRadioPtr.StopSendingCW;
````
was added recently to fix the Escape processing. But we might be ESCaping the CW on the InactiveRadio and NOT the ActiveRadio.

For reference, WinKeyer does not have the issue as it knows where it is sending.

My thought is that we should check if we are sending on both radios OR just stop sending on both Active and Inactive and we hit Escape.

ny4i, 2016-03-17:
This is fixed. It will be merged after Issue149's Pull Request is merged.

Howie- Are you setting the default branch to the latest feature branch. I do wonder if rather than doing the branch for a feature, you could just use a tag. That would keep the number of branches to just real features and not just a place holder branch. Then, you always build the release from master. If you want to go back to a point in time, you checkout a tag with this:  git checkout tags/<tag_name>

n4af, 2016-03-17:
Hi Tom -

I am not following you.
I just pushed up tags v4.47.9 and v4.47.10 for issue 149 and 150, but I do
not see what that is buying us.
Master is presently saying 4.47.6 so I really need to push 4.47.10 just to
sync master ?

73, Howie

http://n4af.net

commit e6439f07 (2016-03-17): Added code to FlushCWBuffer that will stopCW on both Active and Inactive if CWByCAT

Release note: Previously, when sending a CQ on the Inactive radio, hitting the Escape key would not stop sending the CW. This has been corrected.
Closes #150
closed 2016-03-17 by n4af