TR4W/TR4W #151: Pressing F1 with blank call window in S&P strands cursor in Call window

ny4i, 2016-03-17:
If while in S&P mode, you hit F1 with nothing in the call window, your call sign is sent.
When you then enter a call, you cannot get the cursor out of the call window.

ny4i, 2016-03-17:
The issue is that MainUnit.tSetExchWindInitExchangeEntry sets the mweExchange window text to ' ' which is a space. 
````delphi
Windows.ZeroMemory(@ie, SizeOf(ie));
  ie := InitialExchangeEntry(CallWindowString);
//  if ActiveExchange = RSTZoneOrSocietyExchange then ie := ie + ' ';
//  SetWindowText(ExchangeWindowHandle, @ie[1]);
  SetMainWindowText(mweExchange, @ie[1]);
//  if (ie <> '') and (ie <> ' ') then
  if LeaveCursorInCallWindow then tCallWindowSetFocus;
````

The problem is that later on when you type the callsign, the exchange field now has a space in it. When ReturnInSAPOpMode is called,** it tests for ExchangeWindow = '' (nil) NOT space.**

I recommend in ReturnInSAPOpMode that the code check ExchangeWindowString for '' or ' '.

ny4i, 2016-03-17:
This code must be called in many places so this needs to be tested well.

ny4i, 2016-03-17:
I actually found some code in InitialExchangeEntry  in LogEdit that does this:
````delphi
InitialExchangeEntry := ' ' + Exchange; {KK1L: 6.73 Added ' '. K9PG forgets to add it when cursor at start.}
````

This could be the cause of the problem (unintended side effect).

n4af, 2016-03-17:
fixed in 4.47.11 
change tSetExchWindInitExchangeEntry to  
ie := '';     // issue 151 issue151

ny4i, 2016-03-17:
I verified this is working.

ny4i, 2016-03-17:
Release Note: Previously, if F1 was pressed in S&P mode to send your call, when a callsign was entered, the cursor would not go to the exchange window box. This has been corrected.

closed 2016-03-17 by ny4i