TR4W/TR4W #153: CWBC Alt-D swaps radio before sending exchange

n4af, 2016-03-17:
Not sure we have the timing for a correct swap back to run radio, but ALT-D is currently broken using CWBC. There is probably a 'backtoactiveradioafterqso' line that needs to be removed.
Flow:
1) spacebar call station and fills in call window
2) upon pressing enter exchange should be sent. 
3) upon completion of sent exchange, swap radios and
4) initiate cq on active radio

It may not be possible to implement proper timing for #3 but right now STEP 2 is bypassed. and we go directly from step 1 to step 3.

ny4i, 2016-03-17:
If you didn't, can you check with the pull request for 149?

Thanks

Tom

Principal Solutions Architect
Better Software Solutions, Inc.
727-437-2771

n4af, 2016-03-17:
looks like 149 is already merged
u mean #150 ?

http://n4af.net

ny4i, 2016-03-17:
No but that means 149 wasn't the fix. I will look.

Principal Solutions Architect
Better Software Solutions, Inc.
727-437-2771

http://n4af.net

> If you didn't, can you check with the pull request for 149?
>
> Thanks
>
> Tom
>
> Principal Solutions Architect
> Better Software Solutions, Inc.
> 727-437-2771
>
>
> Not sure we have the timing for a correct swap back to run radio, but
ALT-D

n4af, 2016-03-17:
ok. best i can see 149 and 150 are fixed in 4.49.10 (master)

http://n4af.net

ny4i, 2016-03-18:
Here is my latest though on how to make this better:

We know the CW we are sending. We know the speed we are sending it at. Based on the number of CW elements, we can calculate exactly how long it will take to send the characters in question. If we start sending and then set a timer for the time it should wait, the CW should be done when the timer expires.

In the Timer event, we set CWBYCAT_Sending to false and call BackInactive... like we do not in the event in radioPolling.

It is kludgy as hell, but it might work.

n4af, 2016-03-18:
an interesting concept Tom !
Since we now err by over 500ms, I think it will improve things 4 sure.
tks, h

http://n4af.net

ny4i, 2016-03-20:
This is a reference site to where I found some of this information about CW element lengths.

http://sv8gxc.blogspot.com/2010/09/morse-code-101-in-wpm-bw-snr.html

n4af, 2016-03-30:
again experienced this a.m. in CWT using CWBC.
This time i recreated after the end of contest.
Contest = CWOPS
rig 1 80m rig 2 15m
call cq on rig 2 --**WHILE CQ RUNNING** press space to call W9CC who is in ALT-D window.
Program correctly switches over to 80m rig and calls W9CC. I then press enter to log w9cc and send my exchange. Somewhere between two char and no chars sent, CWBC switches back to the run radio.
Since this was created with both K3's in test mode, it cannot be an rfi issue.

ny4i, 2016-03-30:
I will take a look at it today.

Tom

ny4i, 2016-04-19:
Is this still an issue?

n4af, 2016-04-19:
Hi Tom -

Unfortunately so.

73, Howie

http://n4af.net

n4af, 2016-06-06:
There may be an architectural limitation re:CWBC and two radios.
It is best exemplified by pressing an F key, say to call CQ, and then pressing Escape.

Once the string is handed to the firmware, how to stop it ?
Something like ALT-D would be impossible since it is verboten to send on two radios at the same time.

ny4i, 2016-06-06:
Howie,

StopSendingCW in LOGRADIO sends a command to the radio to stop sending. That has worked fine with Kenwood and Icom radios.

Tom
Principal Solutions Architect, ITIL®v3 
Better Software Solutions, Inc.
727-437-BSS1 (727-437-2771)
@BSSI_Consulting

n4af, 2016-06-06:
Hi Tom -

I would think stopsendingcw in ESCAPE_PROC should stop the sending if that
were so ?
But it does not.

73, Howie

http://n4af.net

ny4i, 2016-06-06:
That had been working fine. The CW would stop right away when I hit ESC.
I will have to look.

Tom 
Principal Solutions Architect, ITIL®v3 
Better Software Solutions, Inc.
727-437-BSS1 (727-437-2771)
@BSSI_Consulting

n4af, 2016-06-06:
well, he hits
  AddToOutputBuffer('KY ' + Chr(4) + ';RX;', 8); // Chr(4) stops the K3 //
ny4i Issue 157

but d/n seem to stop the sending.

73, Howie

http://n4af.net

n4af, 2016-06-06:
i am getting fooled by debug mode.
ALT-D seems to work ok under 4.50.
You want to close it ?

n4af, 2016-06-07:
Not sure whether to open a new problem or not, I guess we can continue to use this.
The problem I am seeing with CWBC/ALT-D/F1/ENTER all seems to resolve to speedup char.
Two tests:
1)
Assign to any F key: 
CQ TEST
press escape and the CWBC transmission is successfully stopped
2) Now
assign to any F key
CQ <06>TEST<13>
press escape and you cannot break the sequence
When this is extended to ALT-D, sending occurs on both radios.

n4af, 2016-06-09:
iused cwbc so2r/alt-d heavily in wed CWOps test without problems.

PR #154 (2016-03-18): Fixed issue where Alt-D S&P did not resume CW on Active radio
Release Note: Previously, when using CWByCAT, if Alt-D was used to work the a station, the CQ was not resumed on the Active radio. This has been fixed and closes #153.
commit 1678c27c (2016-03-18): Fixed issue where Alt-D S&P did not resume CW on Active radio

Release Note: Previously, when using CWByCAT, if Alt-D was used to work the a station, the CQ was not resumed on the Active radio. This has been fixed and closes #153.
closed 2016-06-09 by n4af