TR4W/TR4W #160: CWByCAT escape erases initial exchange

n4af, 2016-03-26:
A single escape should stop sending but not wipe out the initial exchange.
Works ok in Wkey.

ny4i, 2016-03-26:
You mean whatever exchange the program pulls up when you enter the call? Such as zone based on call sign? If so, that is an easy fix. I missed that case in my testing. I can do it but if you do, I significantly changed how that Escape code works relative to CWBYCAT. I made it operate more like other modes with the correct state being set by CWBYCAT.

n4af, 2016-03-26:
Hi Tom -

Yes, this is BillWindle QP which prefills name #. I have the Fkey set to
give rst/name/# but the 5NN rst is not always legit and I try to break the
transmission to manually send '579' or whatever- but on CWBC it wipes the
prefill.

Tks, Howie

http://n4af.net

ny4i, 2016-03-30:
In my testing with S&P mode, IF the radio is sending CW, a single escape stops the CW but does NOT clear the initial exchange. If CW is not sending and I hit ESC, the initial exchange is cleared.

I will try WK and the BillWindle QP.

n4af, 2016-03-30:
Hi Tom -

Is this with a K3 radio ?

http://n4af.net

ny4i, 2016-03-30:
I cannot get this to fail. I may need a better handle on exactly what you are doing. Mine works just fine in that if I am sending CW the CW stops.  

WinKeyer works the exact same way.

Yes, with a K3.

n4af, 2016-03-30:
are you loading in a initial.ex prefill ?

http://n4af.net

ny4i, 2016-03-30:
Perhaps not. I have TRMASTER but and that pulls up the name. I worked a station and then on another band, the exchange is prefilled. What is the exchange in this contest?

n4af, 2016-03-30:
i doubt if the problem is contest specific, but for bwqp a single line
reading
9J2BO            BRIAN 1322

in the initial_ex   should cause brian to load,
you need to set custom initial exchange string  to something like 'name
user1'  & i have initial exchange set to CUSTOM
when you enter 9j2bo, his name and number should load into exch window

tks, Howie

http://n4af.net

n4af, 2016-03-30:
Hi Tom -

I just tried to recreate and could not get it to fail.

Let's close it and I will reopen if need be.

Tks, Howie

http://n4af.net

closed 2016-03-30 by n4af