TR4W/TR4W #166: Disableing split on Yaesu radios out of sync

ny4i, 2016-04-17:
UA0DM Victor reported the following:

> Hi dear Howie. 
> II found a strange bug in version 4.4X If I make a double click on the callsign in bandmap, the transceiver records the frequency of VFO A and includes a split mode (VFO B tx). Repeated double click off split.
>  I looked at the data that are transmitted over CAT. After recording the frequency commands in VFOA FAXXXXXXX transmitted commands FR0; FT0 ;. These commands and include the split mode. 
> In earlier versions passed FR0. 
> Apparently FT0 should be removed
> 
> 73! 
> UA0DM Victor

ny4i, 2016-04-17:
According to the manual, FT0; is a toggle for VFO A Band RX On/Off.

FR0; is NOT a toggle and sets VFO A to RX and VFO B RX Off.

When the program wants to turn split OFF, it sends this: FR0;FT0; for the following radios:
FT450, FT950, FT991, FT1200, FT2000, FTDX3000, FTDX5000, FTDX9000:
        

As it looks like FR0; is a toggle, I wonder if it would be more correct to send FR0; FT2;?

 FT2 sets VFO A as the TX. It seems logical that the definition of Not Split is VFO A is both the RX and the TX.

With the program sending the FT0;, it is possible that is the Split is manually disabled, the radio could toggle it back on. If this happened before the program was able to poll and see the status of split, this could cause an issue.

I wonder if anyone with a Yaesu radio has a way to send a command to their radio of FR0;FT2; and verify that disabled split. I do not have a Yaesu rig to check.

If anyone wants to see this code themselves, click here: https://github.com/n4af/TR4W/blob/master/tr4w/src/trdos/LOGRADIO.PAS#L1542

An interesting point, I found that to enable split on these same Yaesu radios, the program sends 'FR0;FT1; . That might have the same toggle issue so possible FR0; FT3 would be better as FT3; explicitly enabled TX on VFO B.

ny4i, 2016-04-17:
I need verification that using_ FR0;FT2;_ will work to **disable **split and_ FR0;FT3;_ will work to **enable **it.

ny4i, 2016-04-18:
More from Victor:

> ok Tom I test it!
> All work !
> 
> 73! Victor
> 
> On Tuesday, April 19, 2016 at 12:12:17 AM UTC+10, NY4I wrote:
> I need Howie’s feedback and then I can proceed.
> 
> I wonder if you were able to test where you send FR0;FT3; to enable split, manually disable split with the radio controls, and then send FR0;FT2; to be certain split is off. Then, go though one more cycle of FR0;FT3; and verify you are in split, then FR0;FT2 to verify you are out of split.
> 
> Thanks,
> 
> Tom NY4I
> 
> 727-437-2771
> 
> 
> 
> On Apr 18, 2016, at 9:36 AM, Victor Titov <92464075...@gmail.com> wrote:
> 
> 
> Hi Tom
> I verifed FR0;FT2; disable SPLIT and FR0;FT3; enable SPLIT !
> The next version will be corrected?
>

n4af, 2016-04-18:
Thanks Tom !

73, Howie

http://n4af.net

PR #167 (2016-04-19): Release Note: Changed Yaesu split On and off to use FT3 and FT2
Howie - Can you build this as the next version but create a beta release so that Victor can try it? Just email the original thread where he can get it. Or let me know and I will pass that on. 

This closes #166 

Thanks
closed 2016-04-19 by n4af